Ultimaker/Cura #20092: Pls add "Show modified settings" button/option to the "Print Settings > Show Custom"

Silencer2079, 2025-01-01:
### Is your feature request related to a problem?

Print settings often need to be changed on a per print basis, and often the same few settings eg. number of walls, surface skin layers, brim width, print speed etc.
While "Show Custom" does highlight settings that have been modified from default, it is currently frustrating to review these as I have to go through each subsection, expand it, pick out those modified settings, collapse the sub section, then do this again for each sub section and there are 14 sub sections. This takes needless time.

### Describe the solution you'd like

A simple checkbox on the "Show Custom" settings interface that, when selected, will filter and display all and only those settings which have been modified from the defaults, without separating and collapsing them in sub sections. Just a simple straight forward list of any non-default setting, I can see them all at once and don't have to do any extra clicking/searching to find anything that is non-default. This would speed up my pre-print workflow considerably. Unselecting this checkbox would return to the normal view.

### Describe alternatives you've considered

N/A - this is a simple UI filter.

### Affected users and/or printers

Potentially every user would benefit from this, at least every user that wants to modify the same settings for each print project.

### Additional information & file uploads

[image]

fieldOfView, 2025-01-01:
The Tabbed Settings plugin adds this as a tab. Check it out in the Marketplace, or here:
https://community.ultimaker.com/topic/43424-tabbed-settings-plugin/#comment-321757
https://github.com/fieldOfView/Cura-TabbedSettingsPlugin/

FabioM3, 2025-01-12:
You can use the plugin Profile analyzer, search in github.

fieldOfView, 2025-01-24:
@Silencer2079, please let us know if any of the two options given work for you.

Silencer2079, 2025-01-30:
Thanks, yes the tabbed settings plugin does exactly what I need.
Thank you.

closed 2025-01-30 by Silencer2079