Ultimaker/Cura #20094: Installing Cura 4.11 or 5.9 on fresh Windows 11, but cannot start program

kirmsi, 2025-01-01:
### Cura Version

4.11 / 5.9

### Operating System

Windows 11

### Printer

Not able to select yet

### Reproduction steps

1. Installed Ultimaker Cura (4.11 and 5.9) via installer on Windows 11. 
2. Trying to start. Program seem to start, but than crashes without error message. 
3. Checking forums and logs and tried to configure 3D settings in Nvidia control panel, but no progress made. 

### Actual results

Trying to start Ultimaker Cura, but after loading, it crashes silently.

### Expected results

I'd like to start the program.

### Add your .zip and screenshots here ⬇️

For the 4.11 installation: 
[cura.log](https://github.com/user-attachments/files/18286820/cura.log) 
For the 5.9 installation:
[cura.log](https://github.com/user-attachments/files/18286822/cura.log)

GregValiant, 2025-01-02:
Thanks for the report.
Have you tried running Cura "As Administrator"?
I'm not very good at reading log files but there are no errors regarding "OpenGL" drivers so the video system may not be a problem.  If you are using more than one monitor - try disconnecting all secondary monitors and try running Cura with a single monitor.

kirmsi, 2025-01-02:
Thanks for your reply - 
I have tried running as administrator - but got the same issues: It seems to start, then it crashes silently. 

I also cannot see anything in particular in the logs, why this shouldn't start. 🫣 
I only have one monitor, so this can't be an issue, right? 

Before the Windows 11 installation - I had running Windows 10 on the same hardware and could use Ultimaker Cura without any problems. 🤔

GregValiant, 2025-01-02:
I'm pretty good at some of this stuff, but when we get into the different operating systems and the video sub-systems I rapidly get out of my depth.
@HellAholic of the Cura Team will take a look at this.

kirmsi, 2025-01-02:
If you need any more information from me, feel free to ask. 😄 

Happy to get some help, I'm really out of ideas. 😢

HellAholic, 2025-01-03:
Might be that the upgrade from win 10 to 11 caused some undesired side effects.
I would say try removing the %appdata/roaming/cura & %appdata/local/cura contents and try to run it again. 
If it still fails I would look at the system event logs to see the crash report by windows, since there is nothing in the log.
That might give you some clue as to the reason you're getting the crash.

kirmsi, 2025-01-03:
I did not upgrade from win10 to 11, I did a fresh installation to a flashed drive. 🤔 

I tried your suggestion and deleted %appdata%/roaming/cura and %appdata%/local/cura, but this didn't help as well. 

But looking into event logs, I can see all the times Cura crashed and saw this:

```Fehlerhafter Anwendungsname: Cura.exe, Version: 4.11.0.0, Zeitstempel: 0x600d5330
Fehlerhafter Modulname: nvoglv64.dll, Version: 32.0.15.6636, Zeitstempel: 0x674f5e6b
Ausnahmecode: 0xc0000005
Fehleroffset: 0x00000000001f0938
Fehlerhafte Prozess-ID: 0x2284
Fehlerhafte Anwendungsstartzeit: 0x1DB5DE89B3C22D0
Fehlerhafter Anwendungspfad: C:\Program Files\Ultimaker Cura 4.11.0\Cura.exe
Fehlerhafter Modulpfad: C:\WINDOWS\System32\DriverStore\FileRepository\nvmdi.inf_amd64_fe5a9cae5c641286\nvoglv64.dll
Berichts-ID: d80f23ba-1cae-4efa-997c-cf633bcf3237
Vollständiger Name des fehlerhaften Pakets: 
Fehlerhafte paketbezogene Anwendungs-ID: ```

So there is a problem with one dll I guess, I tried an "sfc /scannow", a "Dism /Online /Clеanup-Image /RestorеHеalth" which didn't helped at all. 

Afterwards, I forced reinstalled my GPU driver, and now Ultimaker Cura is starting and I can use it. 😄 

So thanks for all the hints. Commenting this with solution for potential other users running into this. 

Closing.

closed 2025-01-03 by kirmsi

GregValiant, 2025-01-03:
Nice job finding and fixing that.