Ultimaker/Cura #20096: Add a new color scheme "fan speed"

yunline, 2025-01-03:
### Is your feature request related to a problem?

Currently users can't know the fan speed of each line until they run the gcode on their printer.

### Describe the solution you'd like

Add a new color scheme "fan speed" to show the fan speed of each line.

### Describe alternatives you've considered

Alternative: Read gcode
Why it's not working out: Gcode is hard to read

### Affected users and/or printers

Everyone who want to tune their fan more accurately will benefit from this.

GregValiant, 2025-01-03:
Thanks for the report.
Have you used the "Advanced Cooling Fan Control" post processor?  You can set the speeds and turn the fan on and off at different layers or for different features (outer-wall, infill, etc.).  You will know exactly what the fan speed is because you set it yourself.  Cura gradually speeds the fan up until the "Regular fan speed at layer" and then leaves it on until the end of the print.

If you open the Gcode in a text editor, you can use the search function and search for "M106".

yunline, 2025-01-03:
Thank you for replying.

> Have you used the "Advanced Cooling Fan Control" post processor? You can set the speeds and turn the fan on and off at different layers or for different features (outer-wall, infill, etc.). You will [...]

Cura changes the fan speed automatically.
For example, when bridging, the fan speed is changed to 100%.
Another example is the setting `Regular/Maximum Fan Speed Threshold`. When the layer time is less than the threshold, the fan speed will be set higher than the "regular speed".

We users want to observe the change of the fan speed, no matter it is set manually or auto generated by Cura. That is why a "fan speed" color scheme is needed.

> If you open the Gcode in a text editor, you can use the search function and search for "M106".

A 3d visualization is much easier to read than text gcode, isn't it?

GregValiant, 2025-01-03:
Gcode is a simple language and you only need to search for the fan command, but yes, it would be easier for someone else to do all the work (and to pay for it) so you would not have to learn gcode.

yunline, 2025-01-03:
> Gcode is a simple language and you only need to search for the fan command, but yes, it would be easier for someone else to do all the work (and to pay for it) so you would not have to learn gcode.

I'm sorry if my tone were not very friendly. I'm still learning English.

I didn't mean to ask anyone that they must implement this feature for me. 

I just want to discuss if adding this feature can make Cura better.

If you don't mind, I can try to open a PR to implement "fan speed" color scheme too.

GregValiant, 2025-01-03:
If you want to write the code for it, I'm sure it will be considered.  Cura is open source and community contributions are appreciated.

GregValiant, 2025-01-03:
Because "Advanced Cooling Fan Control" is a post-processor the gcode would need to be opened in Cura to view the fan speeds.

This is from the Prusa Gcode viewer which does have a fan speed option.  This model was sliced in Cura with Max fan speed of 100% at layer 25.
[image]

This is 100% fan at layer 249.  It's really the same thing, just stretched out over more layers.  My ball bearing 5015 fan wouldn't start to spin until layer 34 as that's when it hits 14%.
[image]

Using the stock Cura fan controls just doesn't show a lot.  It is always a gradual change from 0 to top speed at whatever layer you choose.

This is with Advanced Cooling Fan Control set as "By Layer" and the fan coming on at layer 25 at 100% and turning off at layer 50, then back on at 75% at layer 100 and going to 50% at layer 150.
[image]

This is using "By Feature" with various features at different fan speeds.  Cooling was set to start at layer 50.
[image]

yunline, 2025-01-04:
> Using the stock Cura fan controls just doesn't show a lot. It is always a gradual change from 0 to top speed at whatever layer you choose.

That is not true.

This is my Cura config without any post-processor
[image]
[image]

And this is how the gcode look like in "fan speed" color scheme

[image]

As you can see, the fan speed is changing with the change of the layer time.

[image]

For bridging area, the fan speed is turned to the maximum.

These fan speed changes are generated by cura. They can not be easily predicted by users. A "fan speed" color scheme will make these changes visible. This is why I think it is needed.

LilBub, 2025-01-07:
I agree with his suggestion
Responses like "Cura is open source, go make it yourself" is not the attitude I think we need here.  

There are other slicers out there and to keep Cura on top, we need to embrace suggestions.  Not rebuke them.

mbahler, 2026-02-21:
Agreed. This feature would be great!