Ultimaker/Cura #20097: artifacts

mikeloi2, 2025-01-03:
### Cura Version

5.9.0

### Operating System

windows 10

### Printer

ender 6

### Reproduction steps

 when using cura 5.9 i get artifacts on the printed part

### Actual results

 when using cura 5.9 i get artifacts on the printed part i will be providing a picture of a 25mm cube sliced in cura 5.9 and another with cura 5.8.1, the material is asa the profile is exactly the same

### Expected results

artifacts on the cube

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-01-05:
Thanks for the report.
If you would post both the "Good" gcode and the "Bad" gcode I can compare them.  That may be helpful.
There have been a couple of reports regarding artifacts on curved surfaces (Benchy hull for example) but it would be an odd thing to have those straight lines broken up into several short extrusions that just happen to be in line with each other.  If those shapes are in fact cubes, then each of those extrusions should be a single line from one corner to the other.

mikeloi2, 2025-01-05:
sure thing, here they are ! 
[Desktop.zip](https://github.com/user-attachments/files/18310597/Desktop.zip)

GregValiant, 2025-01-05:
Thank you.

This is from Layer:50 of the the 5.8.1 file.
(The "Line Width" is calculated from the volume and layer height.
The "Time" is calculated from the speed (72mm/sec) and extrusion length).
- Extrusion length------Time------------Flow in mm³/sec--------Line Width
[image]

This is from Layer:50 of the 5.9.0 file.
[image]

Each succeeding wall is 2 "Line Widths" longer, but you can see that there are 4 extrusions in each group.  That's the 4 laps around the cube.  That's the same in both files and from the bottom to the top.  There are no short line segments that would cause a herky-jerky motion which is the primary cause of zitz, blobbing, artifacts, whatever.

Reading both files into AutoCAD - they are like peas in a pod.  I even pasted the 5.9.0 wireframe over the 5.8.1 and they are the same to 8 decimal places.

On every layer, all the wall extrusions go from corner-to-corner with no breaks.  That would seem to indicate that the problem isn't in the gcode.

I'll leave the bug label on this and leave it open, but I can't find anything in there that is pointing to Cura.
Throwing out the thumbnails and the "TIMELAPSE_TAKE_FRAME" lines, the 5.9.0 file is 19 lines longer than the 5.8.1 file.  That's only a 1 line difference every 6 layers or so.  Pretty darn close.

mikeloi2, 2025-01-06:
what mecanical problems can cause this?

GregValiant, 2025-01-07:
If the same printer was used for both, then I have no idea.  I don't print with ASA so I don't have any personal experience with it.
I do print a lot of PETG and it is prone to collecting on the nozzle and dropping off randomly.  That often results in spots on the model, but they are more "stringy" looking - like they got dragged along.  I don't see that on your blotchy model.
Were both models printed from the same roll of filament?  Both with the same nozzle?

Since all the extrusions go from corner to corner, my guess (and it's only a guess) is that something is going on with the printer.
I wish I could be more help.
If you can narrow it down to "it always happens with 5.9.0" then right now my advice would be to use 5.8.x.  At least you would be getting some work done.

Someone from the Cura team will look at this.  Maybe they can shed some light on it as they do a lot of testing with different materials on their printers.