Ultimaker/Cura #20098: Will not Slice

dawaffle77, 2025-01-04:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Ender 3 v3 KE

### Name abnormal settings

it does not slice on default profile

### Describe model location

it does not slice on upon rotating the model in any direction

### Describe your model

I have tried using the meshtool plugin without success it says it is watertight

### Add your .zip here ⬇️

[CE3V3KE_Ruined_Distillery_Floor_0B.zip](https://github.com/user-attachments/files/18308398/CE3V3KE_Ruined_Distillery_Floor_0B.zip)

GregValiant, 2025-01-05:
Thanks for the report.
I didn't have any trouble slicing the project on my Win10 laptop.

Use the _"Help | Export package for technical support"_ then zip that 3mf file and post it here.  Maybe it will have a clue what is happening on your system.

BTW it looks like this:
[image]

dawaffle77, 2025-01-05:
[export-20250105-093209.zip](https://github.com/user-attachments/files/18310710/export-20250105-093209.zip)