Ultimaker/Cura #20099: Slicing a model with settings reset to default for Monoprice Select V1 on cure 5.9.0. Model sliced fine a few times earlier, but never got it to print properly.

Eisenh, 2025-01-04:
### Cura Version

5.9.0

### Operating System

Win 10

### Printer

Monoprice Select Mini V1

### Name abnormal settings

Slicing failed  after resetting print settings to default or any infill less than 60%.

### Describe model location

Centered

### Describe your model

Half-cylinder laying on cut face.

### Add your .zip here ⬇️

[XShot Dart Chain Expansion.zip](https://github.com/user-attachments/files/18308533/XShot.Dart.Chain.Expansion.zip)
Used single link file.

GregValiant, 2025-01-05:
Thanks for the report.
We ask for project files because they contain your settings and printer as well as the model.
This one is kind of weird.  The 'single' stl file has no errors but one of the models in the x10 file does have serious errors.
This is the Cura x-ray view.
[image]

This is how it sliced.  It is missing several layers.  This is with the Infill Density at 5%.  It's possible that the errors in that model may be causing the slice to fail.
[image]

In addition, all the models have gaps where the lettering occurs.  That may be because my settings are different than what you are using.

I was able to repair the x10 file using MS3D Builder.  After repair it looks good EXCEPT at the letters.  I'm thinking they should have been "outies" instead of "innies".
[image]

There have been recent reports of slicing failing on Win11 systems and so I will leave this open.
When a model has errors (like in the x10 file) it's hard to tell if there is a bug in the software, or if the errors are confusing the software.

Eisenh, 2025-02-03:
Thanks for the detailed response. I will look into using the error-free single model.