Ultimaker/Cura #20103: Can't locate place to add custom printer

CakePiper69, 2025-01-07:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

None. Current default printer is Monoprice Mini V2

### Reproduction steps

1. Successfully used Monoprice Mini printer
2. trying to add a Creality CR-10s printer
3. Opened "add printer"
4. selected "non-ultimaker printer"
5. selected "add a non-network printer"
6. my printer is not listed so I opened your page on how to add a custom printer, since I have the settings for this printer
7. went back to "add a printer" section of Cura
8. the option to add a custom printer is not listed (like it says on your website).
9. Tried connecting the USB cable to the printer controller to see if the program would recognize it. No luck
10. Came here to report a bug.  :)

### Actual results

Nothing. I'm still unable to add the new (new to me) printer.

### Expected results

I should be able to add a new printer by make and model or add a custom printer and insert my settings.

PS  I searched for the log file, but was unable to find it.  Sorry.

### Add your .zip and screenshots here ⬇️

[image]

GregValiant, 2025-01-07:
Thanks for the report.
Do you happen to have the "StartUp Optimiser" plugin loaded from the MarketPlace?
One of the things it does is hide printers so Cura doesn't have to load all 300+ definitions.

CakePiper69, 2025-01-07:
Hi Greg,

Thank you for the tip about the startup optimizer. Don't know for certain if that was installed. I have removed Cura from my old laptop (where it worked fine with Windows 10) and from my new laptop running Windows 11 because it was REALLY bogging down boot up. After removing it from the Win 11 laptop, boot up is back to normal.

I may have selected an option on the Win 10 laptop to not load up all the printers to save time, now that you bring that up. Hmmm...  When I installed Cura on the Win 11 laptop, I was able to find and load the Creality CR-10s printer. I'm about to reinstall Cura on the Win 10 laptop and see if I can load the CR-10s there. Really hesitant to install it again on the Win 11 laptop because of what I was experiencing.

Just so you know, I absolutely LOVE Cura. It does everything I need a slicer to do. A wonderful piece of software.

Thank you for responding. Have a great day and a wonderful 2025.

Regards, Walt

T

CakePiper69, 2025-01-08:
Greg,

Thank you so very much for pointing me to the Startup Optimizer. After removing the plugin, I can now see a complete list of all the printers that Cura supports. My day is now complete!  🙂

Must go print stuff now.

Take care, Walt

GregValiant, 2025-01-08:
I'll go ahead and close this.  Have a good day.!

closed 2025-01-08 by GregValiant