Ultimaker/Cura #20110: BB04 Bus with Lid & Cap Failed Slice

brdi, 2025-01-08:
### Cura Version

5.9.0

### Operating System

MacOS BigSur 11.7.10

### Printer

FLSUN V400 brdi

### Name abnormal settings

Just small adjustments. Does not slice with standard profiles, does not slice with other printers nd profiles eg, standard Crealty S1 etc.

### Describe model location

A cylindric jar with perforated lid with cap

### Describe your model

Yes, it slices OK with 5.0.0

### Add your .zip here ⬇️

[FV_BB0.4 Failed Slice 5.9.0.zip](https://github.com/user-attachments/files/18343873/FV_BB0.4.Failed.Slice.5.9.0.zip)

brdi, 2025-01-08:
Added inf regarding the "Model Location": The STL's resides on the harddrive of my Mac

brdi, 2025-01-08:
ReInstalled the FLSUN V400 and tried to slice it with standard Generic PLA profile (no adjustemts). Did not work either.

brdi, 2025-01-08:
[FV_BB0.4 Deksel_Bus_Cap Failed Standard Settings.zip](https://github.com/user-attachments/files/18344033/FV_BB0.4.Deksel_Bus_Cap.Failed.Standard.Settings.zip)
Added the projectfile with standard settings. Models are watertight and analyses shows no problems
[Schermafbeelding 2025-01-08 om 09.13.16.zip](https://github.com/user-attachments/files/18344102/Schermafbeelding.2025-01-08.om.09.13.16.zip)

brdi, 2025-01-08:
Closed it unintentionally

brdi, 2025-01-08:
Closed it unintentionally :-(

brdi, 2025-01-08:
[UCP_BB0.4 Deksel_Bus_Cap Failed Standard Settings.3mf.zip](https://github.com/user-attachments/files/18344183/UCP_BB0.4.Deksel_Bus_Cap.Failed.Standard.Settings.3mf.zip)
On a previous issue I was told that another projectfile was needed. Hence I created a UCP projectfile (File>Save Universal Cura Project) And added that in this comment)

brdi, 2025-01-08:
The CuraTiledInfillLog of today:
dyld: Symbol not found: __ZTTNSt3__114basic_ifstreamIcNS_11char_traitsIcEEEE
  Referenced from: /Users/brdi/Library/Application Support/cura/5.9/plugins/CuraEngineTiledInfill/CuraEngineTiledInfill/x86_64/Darwin/curaengine_plugin_infill_generate (which was built for Mac OS X 12.0)
  Expected in: /usr/lib/libc++.1.dylib
 in /Users/brdi/Library/Application Support/cura/5.9/plugins/CuraEngineTiledInfill/CuraEngineTiledInfill/x86_64/Darwin/curaengine_plugin_infill_generate

GregValiant, 2025-01-11:
Thanks for the report.
I didn't have any trouble slicing either of the projects so I am unable to duplicate the problem (Win 10 Pro laptop).

The Cura team will take a look.

brdi, 2025-01-11:
OK,
Thanks for the info.

I was able to slice the models with 5.9.0  on another machine (iMac Intel 2021) so I guess there is something wrong with the implementation on my MacBook.
Slicing the models on the MacBook ( Intel 2013 so an oldi) with Cura 5.0.0 or 5.7.2 on this machine do not cause any problem…..
Weird.
I don’t understand why the issue started after installing 5.9

brdi

GregValiant, 2025-01-12:
It seems like there are ups and downs to the bug reports for the different operating systems.  It might have something to do with the repairs being made.
The MAC's were having problems and that was fixed.  But then Linux started having problems (but only certain flavors of Linux).  When that was fixed, Windows 11 started having problems.  It seems to be a vicious game of whack-a-mole.
All the patches and updates pushed out by Microsoft and Apple don't help as they can cause unforeseen problems with any installed 3rd party app.

brdi, 2025-01-12:
Clear.I deleted Cura 5.9.0 totally, then reinstalled it reinstalled my printer and filament profiles and now it slices the model as it should. Tnx anywayzBrdi

closed 2025-01-12 by brdi