Ultimaker/Cura #20113: Incorrectly placed Z-seam and pinholes in the surface

DerKob, 2025-01-08:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Artillery SWX1

### Reproduction steps

1. adjustment of the Z-seam to user-defined 

2. component placed exactly in the center of the print bed

3. the Z-seam should be located at the holes inside the object. 
My company has been printing this stl object since 2018

### Actual results

The Z-seam at the holes should be on the inside. Until version 5.8.1, this was possible because you could select the preference of the Z-seam edge in user-defined mode. Since 5.9.0 this is no longer possible. For this reason, the Z-seam at the holes is now on the outside. I have not found a way to move them back to the inside. 

Additionally there is a second problem:

Since version 5, objekts with higher resolution also get small pinholes on the actually smooth surface. 

If you place the object in a different position on the print bed, the pinholes appear on a different place on the surface of the object. 

### Expected results

The Z-seam at the holes should be inside the object and the surface should be free of pinholes.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]

DerKob, 2025-01-08:
Unfortunately, the file cannot be shared because it is the property of my company.

GregValiant, 2025-01-09:
Thanks for the report.
This might be related to changes made to the Z-seam in 5.9.0.  Those changes also appear to have affected the "Layer Start X" and "Layer Start Y" in the Travel Settings.
The Cura team is aware of this bug, and changes are planned for the next release.
In the meantime,with "Z seam Relative" enabled, change the "Z seam Y" to -1 and see if it makes a difference.
This is what I get with my mockup with the "Z seam Y" set to -1.
[image]
This is with your setting of "Z seam Y" at -15.
[image]

I'm not sure about that error highlighted in the blue circle.  My "mockup model" of your model doesn't show that wiggle in the toolpath in 5.9.0. or 5.8.1.
If you can create a model that duplicates that problem, it would be helpful.  You can also get with someone at UltiMaker (@HellAholic ??) and take this off-line.  That is usually acceptable when dealing with proprietary models.

In the meantime, you can slice with 4.13.1 if it is giving acceptable results.  I liked that version a lot.

DerKob, 2025-01-09:
Thanks for the workaround. The -1 setting doesn't create the prettiest Z seam, but it worked. 

I also created a test part to show the problems with the pinholes. If you slice the part with my settings, some pinholes will appear. If you move the part in different places on the print bead, the pinholes will appear in different places on the surface.

Where can I upload the .stl and .3mf files afterwards or to whom can I send them?

DerKob, 2025-01-09:
[Testpart for Pinholes.zip](https://github.com/user-attachments/files/18365488/Testpart.for.Pinholes.zip)
The .stl and .3mf files can be found in this .zip file. You can use them like this. At the moment I have not found any other way to provide the files.

GregValiant, 2025-01-09:
The way it's supposed to work with "Z seam Relative" is that the origin is at the model "Center of Geometry".  Moving the Z seam Y just a little keeps the aiming point for the Z seam within the model.  At -15 it may have fallen outside and so the Z seam was aimed outside.  On the other hand, it may be a bug.
There have been recent changes to locating the Z seam, the addition of the Scarf Zseam, and a couple of other things, that appear to be having "unanticipated consequences".  That is being looked into.

In your "pinhole" file I see this in the preview on the upper part.  (at Z=41.85).
[image]

Reading the gcode into AutoCad, that dip is 0.08mm deep and 0.44mm long.  I didn't find any errors in the model and I don't see anything on the model surface that would cause that.  That does look like a glitch in the toolpath since it makes it into the Gcode.  

There are a lot of these spots where the lines make a quick jig-jog.  The big one is only .02mm but there are a lot of them.
[image]

The Cura team will take a look.

DerKob, 2025-01-09:
Thank you very much for the quick reply. I am very excited about the results and the new Cura version.

HellAholic, 2025-01-09:
@GregValiant The shifts look a lot like the artifacts related to max resolution settings I was working on in 5.4-5.5 version.

@DerKob If you change the max resolution, you can filter them out. Given that you're moving from 4.13.1 to version 5.x a lot of the line behavior has been adjusted with the update to the engine to Arachne.
[image]

There is also the smartavionics version of Cura which is still based on the non-Arachne engine, that version also has a different z seam behavior, so you could also give that one a try, might suit your needs better if you don't want to adjust the max resolution.

smartavionics, 2025-01-09:
See https://github.com/smartavionics/Cura/releases for my Cura variant. As mentioned above, it may be worth trying if you don't need Cura 5 features.

GregValiant, 2025-01-09:
+1 on that.