Ultimaker/Cura #20114: Weird bug causing slicing error

prd0000, 2025-01-09:
### Cura Version

5.9.0

### Operating System

Ubuntu Linux

### Printer

Creality CR-10S

### Reproduction steps

Put the object on the lower right quadrant, and slice. It runs perfectly when you move the object to the left or a bit upper

### Actual results

[image]

### Expected results

[image]

### Add your .zip and screenshots here ⬇️

[BODY ATAS BOTTLE NEW DB-A R1-ALT 1 Mock Up.zip](https://github.com/user-attachments/files/18356044/BODY.ATAS.BOTTLE.NEW.DB-A.R1-ALT.1.Mock.Up.zip)

GregValiant, 2025-01-09:
Thanks for the report.
I can't duplicate this using my settings and printer.  It slices wherever I put it.
Reproduce the problem and then use the "Help | Export package for Technical Support" command.  Zip that 3mf Technical Support file and post the zip folder.  It will include your log file, the model, your printer, and your settings.

prd0000, 2025-01-10:
Here is the file
[export-20250110-115633.zip](https://github.com/user-attachments/files/18370794/export-20250110-115633.zip)

GregValiant, 2025-01-10:
This is one I haven't seen before:
**"Backend exited abnormally with return code -11!"**
I'm not savvy enough with the log files to know what the error code is referring to.
The project still slices for me regardless of the position on the build plate.  I'm unable to get it to "not slice".

The Cura team will take a look.

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant