Ultimaker/Cura #20120: Slice failure of a simple tube shape

lshuhala1, 2025-01-11:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Maker Select 3D V2

### Name abnormal settings

Using the same settings that worked for a dozen other models

### Describe model location

Moving it away from the center does allow it to slice

### Describe your model

It's a new model of a simple tube shape created in Tinkercad

### Add your .zip here ⬇️

[HumidifierHoseAdapter1.zip](https://github.com/user-attachments/files/18386037/HumidifierHoseAdapter1.zip)

GregValiant, 2025-01-11:
Thanks for the report.
It won't slice for me either.  If I just move it away from the midpoint of the bed, and then back again, it slices.
I'll mark this as a duplicate report.  This sort of slicing failure was addressed in 5.7.0.  There seems to be a reversion going on.
From my log file:
Backend exited abnormally with return code 3221225477!
My understanding is that is a generic error code.

The Cura team will take a look.

RlabsLLC, 2025-01-18:
it worked for me with Cura 4.20.25-mb on my computer... maybe try an older slicing software!

GregValiant, 2025-01-18:
There may be something corrupt in that "Good_fine_PLA" profile.  If I re-select that and discard all the changes the model slices.  If I just bring the model in for my Ender it also slices without issue.

lshuhala1, 2025-01-18:
Thanks guys. There are a lot of tweaks in that profile but as I said, I use it often and never had a problem before. This is the first time I've run into a mysterious won't slice issue. Now that I know that moving the model on the bed plane is a possible solution I'll remember that if I run into another occurence.

GregValiant, 2025-01-18:
That model wouldn't slice for me with other profiles, only the "reset" profile.  
Did you happen to any changes to the Prusa I3 definition file?
I ask because that model always slices with my Ender 3 Pro as the active printer.

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant