Ultimaker/Cura #20121: Slower and Slower the longer app is open and the more operations initiated.

J-X-C, 2025-01-11:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality Ender 3

### Reproduction steps

Just using the app. The longer the app is open and/or the more operations performed the longer operations take.  Affects the whole program. If I close and restart the app it runs fine for a while. Eventually I get tired of waiting for something to happen and that is when I restart.  

### Actual results

Just using the app. The longer the app is open and/or the more operations performed the longer operations take.  Affects the whole program. If I close and restart the app it runs fine for a while. Eventually I get tired of waiting for something to happen and that is when I restart.  

### Expected results

I should be able to let it run for days. Loading, slicing, saving etc..

### Add your .zip and screenshots here ⬇️

no issues with printing

GregValiant, 2025-01-11:
Thanks for the report.
I haven't seen this on my own system.  I leave it running for days on this rebuilt Win10Pro laptop and it performs the same at the end as it does at the beginning.
A little more information - Are you printing via the USB or over a network?

The Cura team will take a look.

J-X-C, 2025-01-11:
I save to SD and print from SD.

Robbi62, 2025-01-23:
I can confirm this observation on Cura 5.9.0 and Win10Pro.
Especially when switching between several projects with complex objects I tend to restart Cura to improve the slicing speed. No measurements available, just an observation.