Ultimaker/Cura #20122: IDEX Temps are not correct

Deskwared, 2025-01-11:
### Cura Version

5.9.0

### Operating System

WIndows 11

### Printer

Snapmaker J1s

### Reproduction steps

Select a model needing supports.  Assign material 1 to print the model via the left hotend (T0).  Assing material 2 to print the supports, using the right hotend (T1).  Include a brim.  Ensure that the print temperatures are different and assign standby temperatures lower than print temperatures.

### Actual results

Model is assigned to left extruder and supports are assigned to right extruder, correctly.  Left hotend prints first layer (model) using temperatures for right hotend (supports) and right hotend uses temperatures for left hotend.  After first layer, print temperatures are correct, however, neither hotend responds to "standby" temperatures.

### Expected results

Both hotends should print the first layer with the assigned temperatures.  Both hotends should respond to standby temperatures throughout the course of the print.

### Add your .zip and screenshots here ⬇️

[image]
[image]
[image]
[SJ1_Body2.zip](https://github.com/user-attachments/files/18387495/SJ1_Body2.zip)

Deskwared, 2025-01-12:
With help from the community, I have determined that leaving the brim extruder as “not overridden” is (at least partially) causative of this issue.  Anyway, specifying T0 as the brim extruder corrected the first-layer temp swap issue.  Either the “not overridden” function should be corrected to not swap temps, it should default to whatever nozzle is used to print the outside model walls touching the build plate, or it should be removed altogether and force user specification.

GregValiant, 2025-01-12:
Thanks for the report.
Cura does not directly support IDEX printers.  They can certainly be made to work with Cura, but workarounds are always necessary.
There are two reasons for that.

- Ultimaker doesn't build an IDEX printer and so won't be paying their people to add support to Cura for them.
- All the IDEX printers I've looked at (a lot) are all different in the way they respond to a Cura Gcode.  That makes adding direct support very difficult.

As a result, most of the IDEX machines seem to require post processing.  For example, the Flash Forge machines don't support "G0" and so all the travel moves in a Cura gcode need to be revised.  That includes revising how the travel speed is inserted.

Back to your project, there are a couple of things:
The first is that your "J1s PETG" profile has the "standby temperature" for T0 at 230 and the "standby temperature" for T1 at 190.

You do have the "Material Settings" plugin loaded so go to the _Material Settings_ for each extruder, right-click on the "Standby Temperature" box and select "Use value from material".  That will adjust the standby temperatures.
I think that's the bug here.  The "Material Settings" for Standby Temperature do not seem to import correctly.  In both the Material File for Snapmaker breakaway, and in the configuration file "materials" for Snapmaker breakaway, the Standby Temperature is "150" but it isn't making it into the Material Settings in Cura unless the right-click thing is done.  
[image]

The second thing is that you are printing pretty fast and when Cura takes into consideration the settings "Heat Up Speed" and "Cool Down Speed" (in the "Printer Settings") for the hot ends, there isn't time to drop to the Standby Temperatures.  The layers are only taking 11 seconds and at 2°/sec Heat Up Speed (the default) it would take 42 seconds to heat up from 150 to 235.
```
T0
G92 E0
M109 S205
M104 T1 S193.4 >>> this is after changing the Standby Temperature in the settings to 120°.
```
Another problem is that you may not require the "M109" lines which will almost always incur a 10 second delay before the tool actually goes to work.  It would be necessary to post-process and change all the M109's to M104's.  Unfortunately, the Cura Search and Replace" post processor would change all of them including those in the StartUp Gcode.
I have a pull request in for a re-written "Search and Replace" that includes an option to "Skip StartUp Gcode" and "Skip Ending Gode".  It would replace all the M109's throughout the rest of the gcode.

In closing, when I replace that little model with a big model, and change the Standby Temperatures, the gcode looks as you would expect.  Both hot ends drop to their respective Standby Temperatures do to the longer layer times.
```
T0
G92 E0
M2000 S200 V250 A6000
M105
M104 S245 >>> was an M109 line
M104 T1 S120 >>> standby temperature
```

Deskwared, 2025-01-12:
Thanks for taking a look at it.  I appreciate the effort and time!