Ultimaker/Cura #20127: Build Volume Area not correct for both my printers in Cura 5.9.0

Audiofreek, 2025-01-14:
### Cura Version

5.9.0

### Operating System

MacOS 14.5

### Printer

Elegoo Neptune 3 Pro & Ender 3 S1 Pro

### Reproduction steps

Loaded file as seen in screenshots. Build volume not correct in Cura. Printer settings are correct. Cura limiting my print height to approx. 150 mm. This happens with any tall object. 

[image]
[image]

### Actual results

Cannot slice due to object is taller than build volume.

### Expected results

What should happen is that I am able to slice taller objects. This is not the case as seen in the screen shot. 

### Add your .zip and screenshots here ⬇️

[image]
[image]

GregValiant, 2025-01-14:
Thanks for the report.
Please post a project file ("File | Save Project").  You will need to zip the file and post the zip file directly here and not via email.
Your Gantry Height is 260 so in _One at a Time_ mode the machine will be shorter, but only by 20mm.
When I install the Neptune 3 Pro I get this and it will hold a model 1 layer height below 280.
[image]

My own printer is an Ender 3 Pro and the build height is exactly right as well.

Audiofreek, 2025-01-14:
Thanks for the response. I ended up uninstalling and reinstalling Cura and I left out a bunch of the plugins. That seemed to correct the issue. I am currently printing the Moari project that I posted about.

image.png (view on web)<https://github.com/user-attachments/assets/ac6b7cd8-10dc-4c1f-945f-65fed6d174fd>

GregValiant, 2025-01-14:
I'll go ahead and close this.
If you figure out which plugin was causing that situation, let us know.  The contributors like to know if there are problems with their work so they can fix it.

closed 2025-01-14 by GregValiant

Audiofreek, 2025-01-14:
Will do. Thanks,