Ultimaker/Cura #20133: Z Seam - User Defined - Not operating as expected

stang1966gt, 2025-01-15:
### Cura Version

5.6 5.9

### Operating System

Sonoma OSX

### Printer

CR-10

### Reproduction steps

Imported a standard XYZ cube, centered on build plate.
Changed the User Defined Z Seam location to all 8 positions in the drop down.  Likewise manually entered the Zseam location 50 different ways with no change.  

### Actual results

The ZSeam does not change location from the Front to Back or Left to Right.  It goes where it wants to, in this case stays on the face opposite X but moves along the face.  ZSeam Relative doesn't change the results.   Photos Uploaded.  This issue happens with any STL.  CHEP has a nice video on how the feature used to work at one time.

[image]
[image]
[image]
[image]

### Expected results

The ZSeam should move from one face to the next, when changing User Defined: Front / Back / Right / Left.

### Add your .zip and screenshots here ⬇️

[xyzCalibration_cube.stl.zip](https://github.com/user-attachments/files/18427808/xyzCalibration_cube.stl.zip)
[image]
[image]
[image]
[image]

GregValiant, 2025-01-15:
Thanks for the report.
I know I can duplicate this one.  The Cura team is aware of this bug and is working to fix it.  The "Layer Start X" and "Layer Start Y" are broken as well.  This might have happened when the Scarf Z-seam was introduced.  Hopefully it will be fixed in the next version.
I'll mark this as a duplicate report.

stang1966gt, 2025-01-15:
Thank youI noticed another user reported the issue and update expected for for next release4.13 z seam works correctly?ThanksWill

GregValiant, 2025-01-15:
I believe it was OK in 5.8.0 and 5.8.0.  It has been under construction for a bit though.
I have a model printing now that I couldn't get the seam right in 5.9.0 so I sliced with 5.8.0 and it's fine.

GregValiant, 2025-01-15:
I just looked at your user name.  I had a 70 Mach1 and we currently have an 04 convertible.  The 70 was a beautiful car.  The 04 is a pig.

stang1966gt, 2025-01-15:
Cool, Mach1 are great.  Especially the oval headlight bucket cut outs on the 70They are fun cars but take time, money and space 1992 bought my red 66 coupe 289 auto at 17yrs old, $3,200.  Started restoring it in 1998, left the US in 2001. Still sitting in storage2015 bought a 66 fastback 6cyl that got rebodied, except for a few pieces of metal.  Sitting on a dolly with no suspension but all the parts, epoxy paint.  Also sitting in the same storage facility Thanks for the version feedback. In the mean time, I used 4.13 and it was a dog to slice and the Preview graphics didn’t look as clean as 5.