Ultimaker/Cura #20134: Cura crashing when trying to open or save files

woodwzrd, 2025-01-16:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Creality Ender 3 Max

### Reproduction steps

When I try to open a file or save a file from within Cura, file explore partially opens then everything crashes and closes. I can drag and drop a file into Cura and slice it but as soon as I try to save it everything crashes

### Actual results

xx

### Expected results

xx

### Add your .zip and screenshots here ⬇️

xx

cmjones1234, 2025-01-22:
Same issue here using Linux (Debian 12), except when clicking the OPEN FILE icon or File > New Project or File > Open File(s) or Recent, program closes immediately with no error messages. Same issue when trying to import profile.

woodwzrd, 2025-01-22:
> Same issue here using Linux (Debian 12), except when clicking the OPEN FILE icon or File > New Project or File > Open File(s) or Recent, program closes immediately with no error messages. Same issue when trying to import profile.

Are you using 5.9.0 or an earlier version? I'm trying to narrow things down and figure out if it is a bug with 5.9.0 or if it is an interaction issue with my OS. I'm leaning toward maybe an installation error or just a 5.9.0 bug.

cmjones1234, 2025-01-22:
So far it's happened with 5.9.0 and backwards through at least 5.6.0 (the
last one I tried to no avail). Didn't used to have this issue with 5.6.0
until I tried to update. Once I did, it ceased to load past the "loading"
splash screen and would just shut down. At least it opens but that's as far
as I can get with it.

cmjones1234, 2025-01-22:
I finally got Cura working again in 5.6.0 but then it stopped. Couldn't get
any of them to load so I switched to OrcaSlicer (after finally getting it
to load on Linux). Now Cura 5.9.0 loads up but that's all it does.
I've wiped Linux several times and reloaded with slicers but Cura still
doesn't cooperate.