Ultimaker/Cura #20136: Option for Infill

Bunky101, 2025-01-17:
### Is your feature request related to a problem?

I tend to print at a low infill (10%) when strength is not required. I also find that at a low infill the final layers are not as nice as when the infill is higher (20%). To get around this I modify the mesh and change the infill. This works great except for objects that have odd shapes and are not uniform. 

### Describe the solution you'd like

I would like to suggest a setting that can change the infill so many layers before the first solid layer. For example if a fairly large object, say in the shape of a wave was being printed. To ensure a nice top layer I increase the infill 3 layers before the start of the first top layer. This is hard to do on a complex object and most often you end up with many more layers at a higher infill than needed, or some areas are missed leaving a poorer finish. With an option to increase infill at say 3 layers before the first top layer the slicer will automatically do this where needed but not everywhere. This option will also reduce the effort and time needed to add the meshes and modify the infill. 

### Describe alternatives you've considered

Adding mesh to various areas of the object, changing the infill, slicing, adjusting as necessary, slicing, etc. until getting the desired results. This can be quite time consuming and frustrating depending on the complexity of the object. Infill changes at multiple layers from top or bottom layer would be very beneficial to users.

### Affected users and/or printers

Anyone who uses Cura that is interested in changing the infill at various levels from the top or bottom of an object would get benefits to being able to do this quickly and easily.

GregValiant, 2025-01-17:
Have you tried "Gradual Infill Steps"?  Your request sounds exactly like what it does.

Bunky101, 2025-01-18:
It does sound similar. From the instructions 

 

*	Layer Height / Infill Layer Height = 0.15 mm
*	Infill Density = 40%
*	Gradual Infill Steps = 2
*	Gradual Infill Step Height = 1.5 mm

Then, just below the top surface, the infill will be printed at 40% density for 10 layers. During the next 10 layers, the infill will have 20% density. For the rest of the print (below), the infill density will be 10%.

This leads me to believe that if the object is consistent in height and with some calculations you can get the desired results. If the object isn’t consistent in height you will end up with the potential to have some areas where the infill is quite high to get a reasonable infill to support a lower “top” section. This also seems very complicated solution to simply have two or three layers at a higher infill to support the top layers of an object.

 

Having an option to change the infill at a particular layer or layers based on the layer number from top or bottom would make it very easy to make these changes. While this can be done with mesh changes it is very time consuming as well as, i.e. see picture below.