Ultimaker/Cura #20141: FATAL BUG!!

RlabsLLC, 2025-01-18:
### Cura Version

latest for ARM64 Linux (Well, what pi-apps.io had to offer...)

### Operating System

Linux (Raspberry Pi)

### Printer

Ultimaker S3 (Latest Stable Version) (connected to cloud)

### Reproduction steps

I inserted an object into cura and set supports on it with the secondary filament, and to my surprise, half of the support layers were on layer 1, while the other half was the primary filament!! this same thing happens to me when I try to do a dual-color print! Half layers for the secondary are on the first layer when sliced, while the other half is the primary filament. PLEASE FIX THIS!

### Actual results

the secondary filament (when scliced) will not be half merged on the first layer, nor will the other half be the primary filament

### Expected results

much better slicing outcomes

### Add your .zip and screenshots here ⬇️

https://drive.google.com/file/d/1z_ZGuQBJ7Wxy2mZjB1fBFYNFgqESQpL9/view?t=13 --Video recorded from my Raspberry Pi 5 (Using X11 pages and Simple Screen Recorder)

smartavionics, 2025-01-18:
Hi, that Cura is not an Ultimaker build, it's one of my releases based on Cura 4. For more information about my releases go to https://github.com/smartavionics/Cura/releases. Please open an issue there.

smartavionics, 2025-01-18:
Also, could you please provide a project file that shows the issue. The resulting .3mf file will need to be zipped or renamed to .zip to be able to be uploaded. Thanks.

GregValiant, 2025-01-18:
@RlabsLLC 
I'll go ahead and close this as it doesn't appear to be an "UltiMaker Cura" problem.  If it turns out that it is in fact a UM Cura problem then this can be re-opened.  I hope you understand.

closed 2025-01-18 by GregValiant