Ultimaker/Cura #20143: Bluetooth mouse hangs

rjackr, 2025-01-19:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

E3 with Klipper

### Reproduction steps

Just run Cura

### Actual results

When the program is running, the Bluetooth mouse hangs every now and then, but the touchpad works normally.

### Expected results

The mouse should work normally

### Add your .zip and screenshots here ⬇️

[cura.zip](https://github.com/user-attachments/files/18469724/cura.zip)

rjackr, 2025-01-19:
I found the cause. A Bluetooth device that uses a COM port for communication. In my case, it was wireless headphones. I removed them from the system and the problem disappeared.
I recommend adding an option in Cura something like: Do not use COM ports, or only selected ones.

GregValiant, 2025-01-19:
I have a Feature request ( #19373 ) in requesting to keep USB Printing from being installed by default.  It constantly polls serial devices since it thinks they are all printers.  If it could be installed via a menu command I think things would go much smoother for a lot of people.

I'll make a comment on the Feature Request so that this is linked to it and I'll leave this open.

GregValiant, 2026-09-08:
USB Printing is now disabled by default and this problem with head phones and earbuds should no longer occur.  I'll close this.

closed 2026-09-08 by GregValiant