Ultimaker/Cura #20152: Failure on slicing with specific printer and orientation

ericjray, 2025-01-20:
### Cura Version

5.9.0

### Operating System

Sequoia 15.2 on M2 Mac Mini

### Printer

Fails with Green Machine, succeeds with NewGuy

### Name abnormal settings

No. As a matter of fact, a virtually identical model sliced correctly 10 minutes before the crash

### Describe model location

No on movement. Yes on 90 degrees. 

### Describe your model

Yes, many times. It's water tight. 

### Add your .zip here ⬇️

[Archive.zip](https://github.com/user-attachments/files/18483143/Archive.zip)

GregValiant, 2025-01-21:
Thanks for the report and the files.
This is the second or third report of this behavior.  At least one of the other models was also round and would not slice with the "Standard" profile but would slice with other profiles.
In this case I can get it to slice with "Generic PETG" as the material, but not with "Generic TPU" as the material.

The Cura team (@HellAholic?) will take a look.

HellAholic, 2025-01-21:
Didn't really look into what causes the slice fail on the engine side but a quick model repair resolves the issue. 
Give the fixed model project I attached a try and see if you can slice that one.
[image]

[fixed_model.zip](https://github.com/user-attachments/files/18495688/fixed_model.zip)

GregValiant, 2025-01-22:
I had checked the model with MS 3D Builder and it didn't show any errors.
From [onlinestlrepair.](https://formware.co/OnlineStlRepair):
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
--> 0 Non-Manifold edges  (?)
--> 0 Inverted normals  (?)
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 1834
--> Vertex count changed from 776 to 1844 (+1068)
--> Triangle count changed from 1560 to 4164 (+2604)

It more than doubled the number of vertexes.  That "repaired" model sliced as did yours.