Ultimaker/Cura #20153: Add raplacement patterns in mm/min

FlyingRatBull, 2025-01-21:
### Is your feature request related to a problem?

In my starting G-Code I have multiple G0/G1 moves which use `{speed_travel}` for the feed rate. However the value of `{speed_travel}`is in mm/s whereas Marlin only accepts mm/min thus the moves are incredibly slow.

### Describe the solution you'd like

I'd like to have replacement patterns for speed values like `{speed_travel}` but with mm/min.

For example `{speed_travel_per_min}`.

### Describe alternatives you've considered

* Using expressions to calculate the correct value, i.e. via `{speed_travel} * 60`, but neither Cura nor Marlin support this as far as I know
* Using a constant value, but this would disregard changes to speed values in profiles completely

### Affected users and/or printers

* Everyone should benefit from this
* The repo currently has 38 printer definitions with a speed variable in the start/end scripts in them.

### Additional information & file uploads

* [Marlin G0/G1 documentation](https://marlinfw.org/docs/gcode/G000-G001)

fieldOfView, 2025-01-21:
Cura 5.6 and newer support math expressions in gcode snippets.
The syntax is `{speed_travel * 60}`

FlyingRatBull, 2025-01-21:
That's nice to know @fieldOfView and I just tested it and it works nicely, thanks!
Though:
* The [Wiki page](https://github.com/Ultimaker/Cura/wiki/Start-End-G%E2%80%90Code) doesn't mention anything about expressions
* All the printer profiles use `F{speed_travel}` and should thus use `F{speed_travel * 60}`

fieldOfView, 2025-01-21:
See this section of the wiki: https://github.com/Ultimaker/Cura/wiki/Start-End-G%E2%80%90Code#formulas

Errors in profiles should be reported in a new issue.

closed 2025-01-21 by fieldOfView

FlyingRatBull, 2025-01-21:
Ah, that's why I didn't find it, I searched for "expression" instead of formula 😄.
I will open another issue for the printer profiles.