Ultimaker/Cura #20154: Remember last used directory

FlyingRatBull, 2025-01-21:
### Is your feature request related to a problem?

When using the file dialog to open/save a file, it always uses the default directory (HOME on Linux). Having relevant files in a directory far away from this folder requires browsing through several layers of the filesystem each time you want to open/save something which can be tedious and potentially dangerous if you save something, overwriting it with the expectation that it already is in the right directory but it wasn't.

### Describe the solution you'd like

Remember the directory when using the file dialog and set the next file dialog to start at that directory.

### Describe alternatives you've considered

* Moving all files to HOME which would trash it completely
* Having links to all used folders in HOME which would also fill it up

### Affected users and/or printers

All users who use the file dialog should benefit from this.

GregValiant, 2025-01-21:
The behavior is different in Windows.  Cura remembers the folder where the model file was, and then when you want to save the Gcode, it starts in that folder.  It's pretty much always wrong.

There is a Feature Request ( #19528 ) for additional settings for a default "Models are Here" folder and a default "Gcodes go here" folder ala AutoCad.

KopiasCsaba, 2025-02-14:
Yes, I wanted to report the same. Would save a ton of time upon iterating.