Ultimaker/Cura #20158: Error slicing

Fotrik-cz, 2025-01-21:
### Cura Version

5.9.0

### Operating System

Windows 10 Home 22H2

### Printer

CR10-Smart

### Describe model location

no

### Describe your model

basket walls

### Add your .zip here ⬇️

[CCR10S_plot na 4x2 stěny+vnitrni.zip](https://github.com/user-attachments/files/18489550/CCR10S_plot.na.4x2.steny%2Bvnitrni.zip)
Hello
Slicing failed
I also tried slicing with a different filament -ASA, PLA, etc., the error is still the same. I set the table to arrange automatically, I tried arranging it manually, turning on/off supports, the edge of the mat, but the error is the same. When I removed 2 smaller items, the slicing went well. Then I set the selected items to be multiplied again so that they were all twice and the error appeared again.

GregValiant, 2025-01-21:
Thanks for the report.
I didn't have any trouble slicing the project (Win 10 Pro and Cura 5.9.0).
In Cura, force the slicing error and then use the "Help | Export package for Technical Support" command and zip and attach the 3mf file (not via email).
That 3mf will contain your log file.  Maybe it will have a clue as to what is happening on your system.

[image]

Fotrik-cz, 2025-01-22:
Hello,
Thank you for your prompt reply. I'm sending you the file. Interestingly, when I tried to cut this project with two other instances of Cura open, nothing happened when I pressed the cut button, it said -slicing but not working. I clicked cancel , closed both open Cura instances and then the slicing attempt took place with an error message at the end.

[export-20250122-114558.zip](https://github.com/user-attachments/files/18504798/export-20250122-114558.zip)