Ultimaker/Cura #20164: About Top Ironing

WanderingIceWorld, 2025-01-22:
### Is your feature request related to a problem?

The concentric ironing will cause the excess material to gradually be squeezed to the central area. 

### Describe the solution you'd like

I think if set a options of invert concentric ironing Pattern, it will be better! The concentric ironing will cause the excess material to gradually be squeezed to the central area. But if use zhe ZigZag Pattern sometimes has relatively poor processing behavor in some models (such as cylinders). 
When using the printing mode of "outside first, then inside", it would be better to adopt the ironing mode of "inside first, then outside". There should be such an option available. I think only in this way can the shape tolerance be maintained!
我希望可以设置一个反转同心型烫平走线会更好！使用现有的同心型烫平会导致一些多余的材料被逐渐挤到中心区域，但是使用直线走线的烫平模式在有一些模型下会有比较差的烫平表现（比如圆柱体）。在使用先外后内的打印模式下，也应该使用先内后外的烫平模式会比较好，应该有这样的一个选项。我觉得这样才可以维持形状公差！

### Describe alternatives you've considered

non

### Affected users and/or printers

I think this will affect everyone.

GregValiant, 2025-01-22:
Thanks for the report.
It sounds like your "Ironing Flow" is too high.  What do you have it set to?

WanderingIceWorld, 2025-01-22:
I noticed the problem and set the flow to only 5%. The problem is likely to occur when the quality of the material is not so high(such PETG) or when the top layer is printed, the overflow problem can not be solved by pumpingback (I set 6mm of the pumpingback.