Ultimaker/Cura #20167: Inside to Outside Override

DarkOwl2, 2025-01-22:
### Is your feature request related to a problem?

Inside to outside wall ordering is best for unsupported overhangs, but outside to inside is best for the first layer.

With inside to outside, the first line drawn is separated from the brim, and if it contains sharp corners it can easily pull away.  By printing outside to inside, the brim anchors the first line, and then subsequent lines anchor each other.

To improve success rates, I am currently slicing with both settings and then merging the gcode files manually, so that the first layer is outside to inside, then subsequent layers inside to outside.

### Describe the solution you'd like

Provide a first layer override option for the inside to outside wall setting.

### Describe alternatives you've considered

N/A

### Affected users and/or printers

This should improve printability for any situation where bed adhesion is marginal, eg unheated beds or difficult filaments.

GregValiant, 2025-01-22:
This is an alternative to your workaround.

- Bring in a Support Blocker.
- Scale it to about the size of your build plate in the X and Y and make it "Initial Layer Height" thick.
- Locate it at 0,0,0 on your build plate.
- In the Per Model settings configure it to "Modify settings for overlaps" and as a "Cutting Mesh".
- Use "Select Settings" and then "Wall Ordering" and set it to "Outside to Inside".

You can save that as a project file and it will be ready for you to add a model.

[image]

https://github.com/user-attachments/assets/6078652b-fdb8-4c5e-a5b2-b31f597c4ec4

DarkOwl2, 2025-01-24:
Thanks, that work-around looks useful but I've got lost trying to implement it (despite googling).  I've managed the first three steps but got stuck on the fourth:

>     * In the Per Model settings configure it to "Modify settings for overlaps" and as a "Cutting Mesh".

I don't seem to have the Per Model tool in my sidebar.  I found this:

> Per model settings is a tool in the toolbar on the left. Make sure your sidebar is in the Custom mode, select your object in the viewport, and use the last tool in the toolbar. 

...but I can't work out how to get the sidebar into "Custom mode".

Any help appreciated.

GregValiant, 2025-01-24:
It's this one in your last image.

[image]

DarkOwl2, 2025-01-25:
Further to the above, with thanks to @GregValiant for the pointers, I have refined his method:

The attached file is a Cura project file containing a pre-loaded 100x120x0.2 block assigned as "Mesh Type: Modify settings for overlaps", "Cutting mesh", "Wall Ordering Outside to Inside".

When saved, the project file includes all the Cura settings which would then override the existing settings when the file is loaded.  By opening the project file as a zip archive, I found the "Cura" folder within can be deleted - the result is that, when opened, the block is loaded complete with its cutting mesh properties, but no other settings are loaded so the existing user settings are preserved.

100x120mm is the appropriate size for my printer's build plate (Tina2).  That can be scaled within Cura to suit any other printer (and then re-saved, but note the paragraph above).

0.2mm thickness (Z dimension) is chosen for the following reasons: I found that the override takes effect if it represents at least half of the current layer height.  That means the override will work with any first layer thickness up to 0.4mm, and will not affect a 0.15mm second layer as long as the first layer is no less than 0.125mm.

PLEASE NOTE: This is a neat work-around for the feature request, it does not negate the request itself.

[First Layer Override: Wall Ordering.3mf.zip](https://github.com/user-attachments/files/18546788/First.Layer.Override.Wall.Ordering.3mf.zip)