Ultimaker/Cura #20168: Initial Printing Temperature Not Working

MisterLuMox, 2025-01-22:
### Cura Version

5.9.0

### Operating System

windows 11

### Printer

cr-10 v2

### Reproduction steps

Initial print temperature is not working most of the time. Sometimes when I reboot it works, but usually it does not and I have to manually edit the G code to include it (it's really important!!). 

### Actual results

The temperature always defaults to the Printing Temperature

### Expected results

The first layer print temperature should be at the higher initial print temperature setting.  M104/M109 searches only show 240

[CCR10_lower A arm.zip](https://github.com/user-attachments/files/18510605/CCR10_lower.A.arm.zip)

### Add your .zip and screenshots here ⬇️

[Screenshot 2025-01-22 111337.zip](https://github.com/user-attachments/files/18510609/Screenshot.2025-01-22.111337.zip)

GregValiant, 2025-01-22:
Thanks for the report.
It sounds like the setting you want is "Printing Temperature Initial Layer".

"Initial Print Temperature" is the _"OK you can start now"_ temperature and "Final Print Temperature" is the "Cool down to this before the print ends" temperature.  (My experience is that both of those only work for multi-extruder printers.)

Try just using the "Printing Temperature Initial Layer" setting.  The print will start at that temperature and then just before the start of the second layer you will see the "M104 S" line with the main "Print Temperature" setting.

Let me know if that works for you.

MisterLuMox, 2025-01-22:
Switched to Printing Temperature Initial Layer and that fixed it. The language isn't that clear with 'Initial print temperature'. Adding 'before print starts' might make that more clear. 

Thanks for the speedy response.

GregValiant, 2025-01-23:
Yeah, it isn't good.  I'll see if they are willing to change the tooltips for those.
I've requested a change to the tooltip.
Part of the request is to hide the settings for single extruder machines.  Everything I did pointed to the settings only being used by multi-extruder printers but that may not be true.  See #20172 for the suggested wording.

closed 2025-01-23 by GregValiant