Ultimaker/Cura #20171: [5.9.0] Folder names get mixed up while switching between projects (duplicate of #2201 in CuraEngine )

Robbi62, 2025-01-22:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Creality Ender 6

### Reproduction steps

This is my first issue report, so I hope it is accurate and meets your expectations.
I'm not sure if this issue is based in the core engine or just in the client application. It was initially posted on the Cura Engine issue tracker as [#2201](https://github.com/Ultimaker/CuraEngine/issues/2201).

**Application Version**
CURA 5.9.0

**Binary**
"UltiMaker-Cura-5.9.0-win64-X64.exe"

**Platform**
Windows 10

**Python**
3.8.0 or 3.13.1, as both are installed

**Issue**
While switching between different .3mf projects Cura mixes up folders and it tends to overwrite wrong project and gcodes files.

**My expectation**
Cura keeps the project's folder context for all files to be stored.

I didn't find any similiar report so I'll start from the scratch. There are some steps to go to reproduce this issue.

**Steps to reproduce this issue**
Short summary: create 2 folders, 2 project files, 2 gcode files  and switch between these 2 projects

- create **2 new folders** on any drive, named i.e. "**curaproject-1**" and "**curaproject-2**"
- execute Cura and you'll see an empty Cura desktop

- create a **1st new project**
- add an individual STL file
- save the new project to **[drive:]\curaproject-1\project-1.3mf**
 
- create a **2nd new project**
- add an individual STL file to it
- save the new project to **[drive:]\curaproject-2\project-2.3mf**
 
- close and restart Cura (to be on the same page for reproducing, not really required)

- click file open and Cura will show the 2nd project folder
- select the 2nd project file **[drive:]\curaproject-2\project-2.3mf**
- slice
- Save the current project (i.e. CTRL+S) and Cura will open the **2nd folder** with the correct project file. You will be asked to overwrite the existing project - say YES.
- Save the **2nd GCODE** and Cura will open the correct (2nd) folder - **[drive:]\curaproject-2\project-2.gcode**

Let's switch the project:
- click file open and Cura will show the **2nd project folder**, the one we stored our files into right before
- select the **1st project folder** and open the 1st project file **[drive:]\curaproject-1\project-1.3mf**
- slice

**=> Everything is fine until now.**

Now we will run into troubles, as Cura looses its "folder context".
Keep in mind, we are working on **project-1**, opened from folder **[drive:]\curaproject-1\project-1.3mf** 

**=>1st issue**
- Save the current project (i.e. CTRL+S, **project-1.3mf**) and Cura will open the wrong **2nd! folder**
- Correct the folder, save the project file and accept the overwriting of the elder 1st project file **[drive:]\curaproject-1\project-1.3mf**

**=> 2nd issue**
- Save the GCODE and Cura will open the wrong (2nd!) folder.
- Correct the folder, save the 1st GCODEfile **[drive:]\curaproject-1\project-1.gcode**

Let's switch the project for the 3rd time:
- click file open and Cura will show the 1st project folder, the one we stored our files into right before
- select the 2nd project folder and open the 2nd project file **[drive:]\curaproject-2\project-2.3mf**
- slice again

Keep in mind, we are now working on **project-2**, opened from folder **[drive:]\curaproject-2\project-2.3mf**

**=> 3rd issue**
- Save the current project (i.e. CTRL+S, **project-2.3mf**) and Cura will open the wrong **1st! folder**
- Correct the folder, save the project file and accept the overwriting of the elder 2nd project file **[drive:]\curaproject-2\project-2.3mf**
- Save the GCODE and Cura will open the correct (2nd) folder.

**Summary**
Cura doesn't keep the folder context of the current project we are working on when storing files. It tends to switch over to a prior project folder. This will lead into a catastrophic failure if there are files with the same filenames in both folders!

The latter has been avoided in my above example by naming all files with different names. In real life it will happen, that you overwrite i.e. different versions of the same project stored in different folders.

### Actual results

When switching between different .3mf projects, Cura mixes folders, **which can inadvertently destroy files**.

### Expected results

Cura keeps the project's folder context for all files of an active project.

### Add your .zip and screenshots here ⬇️

No files added.

GregValiant, 2025-01-23:
Thanks for the report.
This was previously reported in #11756, #11803, #13095.  I thought this had been addressed.
The Cura team will take a look.

Robbi62, 2025-01-23:
Hi Greg,
As result of investigating into this issue, I could identify the more critical issue #20175, that overwrites wrong project files used before.

I assume, #20175 is the more generic issue to be addressed. If fixed, this as well as the other mentioned issues may be solved as well.