Ultimaker/Cura #20173: [5.9.0][5.8.0][5.7.0] Silently crashing on startup. Faulting module ucrtbase.dll and exception code 0xc0000409

SlickStretch, 2025-01-23:
### Cura Version

5.9.0

### Operating System

Windows 10

### Printer

Creality Ender 3

### Reproduction steps

1. Install Cura 5.9.0, 5.8.0, or 5.7.0
2. Launch Cura
3. Accept the terms
4. Sign in to an Ultimaker account OR click 'skip'

### Actual results

Program closes with no error message.

Windows Event Viewer shows:
```
Faulting application name: CuraEngine.exe, version: 1.0.0.0, time stamp: 0x660c03e2
Faulting module name: ucrtbase.dll, version: 10.0.19041.3636, time stamp: 0x81cf5d89
Exception code: 0xc0000409
Fault offset: 0x000000000007286e
Faulting process id: 0x2790
Faulting application start time: 0x01db6d5c7abba5fa
Faulting application path: C:\Program Files\UltiMaker Cura 5.7.0\CuraEngine.exe
Faulting module path: C:\Windows\System32\ucrtbase.dll
Report Id: 2ddd0127-8ebd-48f1-b465-6c7a57e17682
Faulting package full name: 
Faulting package-relative application ID: 
```

### Expected results

Cura setup should continue and show printer configuration.

### Add your .zip and screenshots here ⬇️

So far I have tried the following, none of it made any difference:

- Delete cura folder from both Appdata\Roaming and Appdata\Local directories.
- Ran `sfc /scannow` in the Windows terminal.
- Install various versions of Cura. Each time uninstalling and deleting the above folders. Version 4.6.0 is the latest one that works.
- Clean re-install of display drivers
- Disconnect additional displays

[Cura 5.9.0 log](https://github.com/user-attachments/files/18515981/cura-5.9.0.log)

[Cura 5.8.1 log](https://github.com/user-attachments/files/18516050/cura.log)

[Cura 5.7.0 log](https://github.com/user-attachments/files/18516068/cura.log)

GregValiant, 2025-01-23:
Thanks for the report.
This problem seems to be specific to your system as there haven't been any other reports like this.  Tell us more about the system as the Cura team will need more information.

As I recall, there were some conflicts when apps like "Chitubox" were installed.  I also seem to recall some other conflicts (Microsoft Widgets?) as well.  I'm running Win10Pro and have about 10 versions of Cura installed and there has never been a problem.

SlickStretch, 2025-01-23:
> more about the system as the Cura team will need more information.

I don't know what's relevant. So, here's all of the info about my system:

```
Summary
		Operating System
			Windows 10 IoT Enterprise LTSC 64-bit
		CPU
			AMD Ryzen 7 3800X	71 °C
			Matisse 7nm Technology
		RAM
			48.0GB Dual-Channel DDR4 @ 1319MHz (16-15-15-35)
		Motherboard
			Micro-Star International Co., Ltd. B550M PRO-VDH WIFI [CEC] (MS-7C95) (AM4)	52 °C
		Graphics
			E420i-A0 (1920x1080@60Hz)
			DELL U2312HM (768x1366@60Hz)
			AS193i (1280x1024@60Hz)
			4095MB NVIDIA GeForce GTX 1080 (MSI)	60 °C
		Storage
			1863GB Seagate ST2000DM001-9YN164 (SATA )	50 °C
			1863GB Hitachi HMRSK0002TBAT07K (SATA )	45 °C
			1863GB Seagate ST2000DM006-2DM164 (SATA )	46 °C
			476GB TEAM T253512GB (SATA (SSD))	33 °C
			447GB KINGSTON SA400S37480G (SATA-2 (SSD))	61 °C
			465GB TOSHIBA MQ01ABF050 (SATA )	42 °C
			931GB Western Digital WD Elements 1078 USB Device (USB (SATA) )	50 °C
		Optical Drives
			PIONEER BD-RW BDR-TD04 USB Device
		Audio
			NVIDIA Virtual Audio Device (Wave Extensible) (WDM)
Operating System
	Windows 10 IoT Enterprise LTSC 64-bit
	Computer type: Virtual
	Installation Date: 11/15/2024 2:31:34 PM
	Serial Number: RV6T3-893K4-7H3T9-VFD8Y-FGGBV
		Windows Security Center
			User Account Control (UAC)	Disabled
			Firewall	Enabled
		Windows Update
			AutoUpdate	Not configured
		Windows Defender
			Windows Defender	Enabled
		Antivirus
			Antivirus	Enabled
			Display Name	Windows Defender
			Virus Signature Database	Up to date
		.NET Frameworks installed
			v4.8 Full
			v4.8 Client
			v3.5 SP1
			v3.0 SP2
			v2.0 SP2
		Internet Explorer
			Version	11.3636.19041.0
		PowerShell
			Version	5.1.19041.1
		Java
				Java Runtime Environment
					Path	C:\Program Files\Java\jre1.8.0_441\bin\java.exe
					Version	8.0
					Update	441
					Build	07
		Environment Variables
			USERPROFILE	C:\Users\Stretch
			SystemRoot	C:\Windows
				User Variables
					Path	C:\Users\Stretch\AppData\Local\Microsoft\WindowsApps
					C:\Users\Stretch\.dotnet\tools
[... 5288 lines cut ...]
				Wi-Fi (Hide Yo Kids Hide Yo Wifi)
					SSID	Hide Yo Kids Hide Yo Wifi
					Frequency	5180000 kHz
					Channel Number	36
					Name	Hide Yo Kids Hide Yo Wifi
					Signal Strength/Quality	93
					Security	Enabled
					State	The interface is not connected to any network
					Dot11 Type	Infrastructure BSS network
					Network	Connectible
					Network Flags	There is a profile for this network
					Cipher Algorithm to be used when joining this network	AES-CCMP algorithm
					Default Auth used to join this network for the first time	802.11i RSNA algorithm that uses PSK
				Wi-Fi (Xfinity Mobile)
					SSID	Xfinity Mobile
					Frequency	5785000 kHz
					Channel Number	157
					Name	Xfinity Mobile
					Signal Strength/Quality	77
					Security	Enabled
					State	The interface is not connected to any network
					Dot11 Type	Infrastructure BSS network
					Network	Connectible
					Network Flags	There is a profile for this network
					Cipher Algori

GregValiant, 2025-01-24:
That may or may not be overkill.  I'm positive about that.