Ultimaker/Cura #20175: [5.9.0] downto [4.13.1] Project names get mixed up while switching between projects

Robbi62, 2025-01-23:
### Cura Version

[5.9.0 Win10] downto [4.13.1 Win7] 

### Operating System

Windows 10 down to Windows 7

### Printer

Creality Ender 6

### Reproduction steps

2nd post after intensively testing issue [#20171](https://github.com/Ultimaker/Cura/issues/20171) on an elder Cura version on Windows 7.

The current issue is more generic and IMHO can produce catastrophic results due to overwritten (project) files. After identifying this issue on the current version of Cura due to already lost project files, I went back to an elder Cura version on Windows 7 with a similiar behaviour. 

Steps to reproduce this issue
Short summary: create 2 folders, 2 project files, 2 gcode files and switch between these 2 projects

- create 2 new folders on any drive, named i.e. "cura_cube" and "cura_cylinder"
- execute Cura and you'll see an empty Cura desktop

**Cura 4.13.1 on Win 7 Ultimate**
- create a 1st new project
- add an individual STL file, i.e. a **calibration cube**
- slice 
- save the new **cube gcode to [drive:]\cura_cube\cube.gcode**
- save the new project to [drive:]\cura_cube\cube.3mf
- create a 2nd new project
- add an individual STL file to it, i.e. a **calibration cylinder**
- slice
- save the new **cylinder gcode to [drive:]\cura_cylinder\cylinder.gcode**
	
### Cura 4.13.1 Issue:
**Cura creates/overwrites the project name**, in this case "**cube**" instead of **cylinder**!
- save the 2nd project
- the selected folder is [drive:]\cura_cube\ [WRONG]
- the selected **filename is cube.3mf** [WRONG]

Both file/folder suggestions are wrong, we are working in the context of cylinder!

- 

**With Cura 5.9.0 we need some additional steps to reproduce this issue**

- create a 1st new project
- add an individual STL file, i.e. a **calibration cube**
- slice 
- save the new **cube gcode to [drive:]\cura_cube\cube.gcode**
- save the new cube project to [drive:]\cura_cube\cube.3mf
- create a 2nd new project
- add an individual STL file to it, i.e. a **calibration cylinder**
- slice
- save the new **cylinder gcode to [drive:]\cura_cylinder\cylinder.gcode**
- save the new cylnder project to [drive:]\cura_cylinder\cylinder.3mf

**- reopen the first **Cube** project [drive:]\cura_cube\cube.3mf**
- slice 
- save the new **cube gcode to [drive:]\cura_cube\cube.gcode**, overwrite the existing gcode file
	
### Cura 5.9.0 issue:
**Cura overwrites the project name**, in this case with **cylinder** instead of **cube**, to be seen directly on the desktop!
- save the 2nd project
- the selected folder for our **cube** project is [drive:]\cura_cube\ [OK]
- the selected filename for our **cube** project is **cylinder.3mf** [WRONG]

**This project name is wrong, we are working in the context of cube!**

**Important**
The behaviour is depending of what will be saved as first.
The above descriptions require to save a sliced .gcode file BEFORE saving the corresponding project file.

### Actual results

  When switching between different .3mf projects, Cura renames project files when saving gcode files before saving project files.

### Expected results

Cura keeps the project's filename and folder context for all files of an active project.

### Add your .zip and screenshots here ⬇️

Cura 5.9.0 on Win 10 Pro

Create and slice the new project 1 "cube"
[image]

Save the new gcode file
[image]
[image]

Save the new project file cube3mf
[image]
[image]

Create and slice the new project 2 "cylinder"
[image]

Save the new gcode file
[image]
[image]

Save the new project file cylinder3mf
[image]
[image]

Reopen and slice project 1 "cube"
[image]
[image]

Before storing the cube gcode
[image]

After storing the cube gcode, the project got renamed to cylinder!
[image]

rburema, 2025-02-05:
I can confirm this is happening -- for 5.9 (at least) I don't have to go through the creation of the cube/cylinder each time, but I can keep loading the project files instead and slicing in-between and it'll also reproduce.

Robbi62, 2025-09-24:
As Cura version 5.10.0 has solved this issue, the status of this issue should be updated.

closed 2025-09-24 by rburema