Ultimaker/Cura #20177: Slicing failed octopus

sporksofdoom, 2025-01-23:
### Cura Version

5.9.0

### Operating System

macOS Sonoma 14.5

### Printer

Elegoo Neptune 3 Pro

### Describe your model

Flexi Octopus. Have sliced it before

### Add your .zip here ⬇️

[EN3PRO_Octopus_spiral_v6_bug.3mf.zip](https://github.com/user-attachments/files/18524680/EN3PRO_Octopus_spiral_v6_bug.3mf.zip)

GregValiant, 2025-01-24:
Thanks for the report.
The project would not slice for me either.  The model does have some errors in it.  Even after repairing the model it still would not slice in 5.9.0.
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
**--> 16 Non-Manifold edges  (?)
--> 32 Inverted normals  (?)**
--> 0 Duplicate faces  (?)
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 24464
**--> Vertex count changed from 230051 to 229442 (-609)
--> Triangle count changed from 460258 to 459280 (-978)**

The Cura team will take a look.

GregValiant, 2026-09-08:
I'll close this.

closed 2026-09-08 by GregValiant