Ultimaker/Cura #20180: Printing order with multiple objects

yamaha1100yzf, 2025-01-24:
### Is your feature request related to a problem?

I want to select printing order when I print multiple object. Even I use "print one at a time" or "print all" . 

### Describe the solution you'd like

I believe, if we can select printing order, this problem will be solved

### Describe alternatives you've considered

There is no alternative. it can be placed in the menu as a choice

### Affected users and/or printers

it will affect and improve the bed adhesion

GregValiant, 2025-01-24:
I think we need a little clarification here.
"Print Ordering" is an option for "One at a Time" print sequence, but not for "All at Once".
Is this Feature Request meant to expand the print ordering to "All at Once"?

In "One at a Time", when "Set print sequence manually" is checked ---
[image]

Right-clicking on the object list brings up the ordering options ---
[image]

yamaha1100yzf, 2025-01-27:
@GregValiant Thanks for your valuable reply. I know this feature but I dont want to use "one at a time" , I want to use "all at once"

yamaha1100yzf, 2025-01-27:
@GregValiant 
Unfortunately, these settings do not work for me. Let me explain this with an image. The image shows the first layer view of the parts I will print. In the image, there are two side parts and five main parts in the middle.

As seen in the image, Cura first prints part number 1 on the left, then part number 4 in the middle. After that, it prints parts 5, 6, 3, 2 in sequence, and finally, part number 7 on the right.

However, I want the printing order to be 1-7-2-3-4-5-6 so that I can apply adhesive to the print bed sequentially for each part as the first layer is being printed.

[image]

GregValiant, 2025-01-27:
It looks like 1 and 7 are being bonded to the other parts.  Can you just export the entire build plate contents from Cura as an STL and then open that file in Cura?.  You should get a single model laid out as you want.

The problem I see with the feature request is that it is a "niche" request.  The chances of it being picked up by the Cura Team and acted on within the next couple of years is not good.  The chances of a "member of the community" picking up on this and taking on massive changes to the Cura code for one particular project is really poor.

If you can figure out how to configure the model's themselves, or if you can open the gcode and re-order models 1 and 7 to the beginning of each layer, that's probably the best you can do in any reasonable time frame.