Ultimaker/Cura #20184: cannot slice even deleting some files

Mickji, 2025-01-24:
### Cura Version

5.6.0

### Operating System

Windows 10

### Printer

Elegoo Neptune 3

### Name abnormal settings

I use 100% infill

### Describe model location

no

### Describe your model

I use 100% infil

### Add your .zip here ⬇️

the program run in an unexpeted error without a number. trying to move the files made impossible finishing the slicing process. I cannot save the file because it gaves error layer height

[Desktop.zip](https://github.com/user-attachments/files/18539051/Desktop.zip)

GregValiant, 2025-01-24:
Thanks for the report.
Cura 5.5 and 5.6 had well-documented problems slicing certain models.  I suggest you update to Cura 5.9.0 as the slicing problems appear to have been fixed.

This is in 5.9.0.  I changed the line width and nozzle size to 0.20 and those small models simply don't slice correctly.  The wall at the flat end is only 0.01mm thick.  You would need a .01 nozzle to slice them.

[image]

closed 2025-01-24 by GregValiant

Mickji, 2025-01-25:
Hi and thank your for your reply, I have the default nozzle of the Neptune 3
I've tried updating the version to 5.9
I am curious, is there a different between elegoo cura and ultimaker cura?
I've also tried deleting the small cilinders pegs from the plate but it gives the same error
About the duplicate and close I hope it will not cause something bad against my account. If I've made an error I wish to ask what I should do to fix it because I've never used these sites. I wrote here because the program was telling me to do so XD

GregValiant, 2025-01-25:
No, the versions are not the same.

It looks like Elegoo forked from UM Cura 5.6 and they now maintain their own version.  Creality and Sovol did the same sort of thing.  Ultimaker Cura is no longer being supported by those companies.
Because of that, they are not submitting any new printer definitions to UM and they no longer update their existing printer definitions here in UM Cura.  The definitions that were submitted will likely stay, but as UM Cura advances - there are new settings being introduced, and other settings get deleted while Elegoo Cura may be going a different way.  The existing printer definition files won't take setting differences into account.  Eventually the definition files will go from simply being "out of date" to "buggy".

If I notice that a bug report has come up previously;  I cross-reference the current report with the original report and then close the duplicate.  That keeps this site from having a bunch of repeats regarding a single problem.  I still get responses from closed reports and I'm usually pretty good about responding (unless I'm out fishing of course).

I don't know how Elegoo handles their bug reports or if they even have a system.  I also don't know if they have made any changes to their version since they forked it.  5.6 really didn't work very well.

GregValiant, 2025-01-25:
I just looked at your project again.  I'm running UM Cura 5.9.0 on a Win10Pro system.
I threw out all those little tube looking things as they aren't really printable.  The rest of the slice with the blocks and the curved piece looks good.