Ultimaker/Cura #20186: PauseAtHeight plugin misses one layer when setting Pause at Height

NitemareReal, 2025-01-25:
### Cura Version

5.9.0

### Operating System

Windows 11

### Printer

Anet A8 Plus

### Reproduction steps

1. Open any supported model file (stl for example)
2. Configure print options, set "First Layer Height" and "Layer height" to 0.2
3. Go to menu "Extensions" -> "Postprocessing" -> "Modify GCode"
4. Press "Add command sequence" button and select "Pause at height"
5. Select Pause at "Height"
6. Set "Pause height" to 5.0
7. Select Method "Marlin (M0)"
8. Mark "Park Print"
9. Set any coordenates, for example X = 10 Y = 250
10. DO NOT check "Redo Layer"
11. Check "Use M109 for standby..."
12. Set standby Temperature (for example 220ºC)
13. Leave all other options unchagned

### Actual results

When printing, just before printing layer 23th (Z = 4.8), printing process pauses, and after resume, it resumes at height = 5, so, it prints layer 23 (Z=4.8) at Z=5 and then it prints later 24 at (again) Z=5

If you see attached screenshot with GCode result, you can see:

1. Z goes to 4.8 (layer 23)
2. Plugin "believes" it is at Z=5
3. Plugin increase Z one milimeter (Z=6)
4. Plugin increase Z to 15 milimeters
5. Pause
6. Z go back to 5 milimeters (wrong, but plugin believes it is right)
7. Layer 23 is printed at Z=5, but Z should be 4.8, as seen in point 1

### Expected results

Pluging code should be injected just before priting layer 24 instead of layer 23. If you set **Pause at layer**, it works ok

### Add your .zip and screenshots here ⬇️

[cubo.zip](https://github.com/user-attachments/files/18547699/cubo.zip)

[image]

[image]

GregValiant, 2025-01-25:
Thanks for the report.
I can confirm this bug.
IMO "By Height" has always been buggy.  It doesn't work with either Adaptive Layers or if Z-hops are enabled.  The fact that it took so long to find this bug is a testimony to the fact that not many people use "By Height".
If you use "By Layer" it should be fine.

I have done a re-write (but I haven't submitted it yet) and as part of that, I scrapped "By Height" and only "By Layer" is left.
Here is "PauseAtLayer".  If you want to test it, unzip the file and place the .py file into the "scripts" sub-folder within your Configuration Folder.  It will be available with the other post processors.
I've been using it for about a year and haven't found any bugs in the last 4 or 5 months.

[PauseAtLayer.zip](https://github.com/user-attachments/files/18548379/PauseAtLayer.zip)

NitemareReal, 2025-01-27:
Thankyou so much. I'm testing it!

closed 2025-01-27 by NitemareReal