Ultimaker/Cura #20187: material section grayed out when selection Ultimaker 2  and 2 extended as printeres, it will not calculate print weight

FastZombieLLC, 2025-01-25:
### Cura Version

5.9

### Operating System

windows 11

### Printer

ultimaker 2 extended

### Reproduction steps

use any cura version im using 5.9, 
select printer to be UM2 extended
go to the preference and try to import or select a material.
You cant all is grayed out.

This this critical to me as I need to know how much the part will weigh when done, I'm working on an ulta light 3d printed product.
Thanks

### Actual results

?

### Expected results

?

### Add your .zip and screenshots here ⬇️

?

GregValiant, 2025-01-26:
Thanks for the report.
I can't reproduce this and so we will need a project file.
Use the "Help | Export file for Technical Support" command and zip that file and post it here (not via email).

HellAholic, 2025-01-27:
As far as I know the material and profiles are handled by the Firmware on the UM2+Extended. That's the reason the material option is greyed out. So you can enable the has_materials in the definition file to get a rough estimate but it's the machine's Firmware that takes over the material settings and the weight of the final print is linked with that.

FastZombieLLC, 2025-02-05:
@GregValiant Thank you very much, this fixed my issue, and I can finally see the weight of the build.
Much appreciated :)

GregValiant, 2025-02-05:
Glad it helped.
I'm going to go ahead and close this as it isn't a bug, but rather how the UM printers work.  The fix can be done on an individual basis for those who need it.

fieldOfView, 2025-02-06:
If you want to handle the materials in Cura instead of on the UM2, you can set the gcode flavor to Marlin (instead of Ultimaker 2). Then you have full control over the material in Cura, and what you set in Cura will override what is set in the printer.