Ultimaker/Cura #20188: Random Corner Seams

NekarXenos, 2025-01-26:
### Is your feature request related to a problem?

Setting seams to random creates bumps everywhere. Setting it to sharpest corner makes one corner sharper than the others if you are printing a box.

### Describe the solution you'd like

What I would like to see is an option for random corner seams, where it makes the seam at a new corner on every layer.

### Describe alternatives you've considered

alternatives are random seams, custom seams or sharpest corner.

### Affected users and/or printers

It would affect all printers as I'm sure it would help everybody

BarsMonster, 2025-01-26:
I personally like random seam as well, but dream to have an option "not closer than x mm than previous layer" or something like this. Very often random seams match each other for multiple layers and become very visible.

Your proposal looks like "clock-wise seam selection", where each seam should be at least x° (for example, 10°) away from seam location in previous layer. This will make seam circle around the part in systematic way for simpler geometrical prints.

Asterchades, 2025-01-26:
I don't understand how this is appreciably different to the existing random seam location option. Random will only ever put the seam on a corner (*Z Seam on Vertex* is only an option for the *Shortest* or *User Specified* seam locations - both *Random* and *Sharpest Corner* enforce it), with that corner changing on every layer - exactly what you're proposing a feature should be added to do. I do know there's some issues with the random placement at the moment which causes it to favour one direction on the model but that's already being looked into.

Adding a minimum distance or rotational offset does at least sound like additional functionality, though I question whether it will be beneficial when the random locations have been corrected.

BarsMonster, 2025-01-27:
I've just tested and there are still issues with random seam placement in 5.9.0: https://github.com/Ultimaker/Cura/issues/8464

Even if it was truly random - sometimes seam will randomly match on sequential layers which will make it more visible. So some way to enforce non-0 distance between seams on sequential layers would help alot.

NekarXenos, 2025-01-28:
Rethiniking it, yes clockwise might be better than random.