Ultimaker/Cura #20189: Simplified brick layers support feature

BarsMonster, 2025-01-26:
### Is your feature request related to a problem?

Brick layers are a new way to 3d print to improve layer adhesion & strength of 3d printed parts. Full-featured request is #18353 , but I afraid it might take quite a bit of time to see it implemented in Cura, as printing in half-layers might require substantial code changes. 

### Describe the solution you'd like

I propose simplified implementation of brick layers, which **might** give 70% benefits while probably being 95% easier to implement:

Add configuration option to separately configure outer perimeter line thickness for odd and even layers separately.

For example, for 0.4mm nozzle, for odd layers outer perimeter line thickness is 0.4mm, and for even - 0.6mm. This way sequential layers will not match each other spatially and we might expect some adhesion improvement. 

### Describe alternatives you've considered

Full featured implementation using half-layers. 

### Affected users and/or printers

All 3d printers should benefit from it. As it's an optional feature - it should cause no degradation.

GregValiant, 2025-01-26:
I like it.  These are some thoughts.

In your example of .4 line width and .6 line width - It looks like the 0.2 additional overlap on the walls would be sufficient.  The layer above traps the inner wall of the layer below.  Sort of like "Alternate Extra Wall" works on the infill.

I have no idea how difficult it would be to add something like this (changing the outer wall line width and moving the toolpath every other layer).  One thing that occurred to me is that there will be an overhang on the inside every-other-layer and so to maintain the "Infill overlap" the infill would need to be adjusted as well.  Depending on how the code is written in Cura Engine, that may or may not be something that already happens.  Currently, the "stop point" for the infill must adjust depending on the outer wall of the model as the inner edge of the inner walls would always be parallel to the model outer wall.  This feature request changes that layer-by-layer.

I scribbled up the cartoon below.  I think it's what we are talking about here.

- The bottom is the current method.
- The middle is with the current Infill Overlap parallel to the model outer wall and is without an adjustment.  It looks like it would leave the yellow areas as voids.  The infill wouldn't have anything to stick to and might fail every other layer.
- The top shows the wide outer walls and the infill is adjusted so there are no voids.  I angled the outer model wall just to see what it would look like.  It makes the alternating overhangs worse, but there would be overhangs anyway.

[image]

BarsMonster, 2025-01-26:
@GregValiant Yes, top picture is something I had in my mind as well. 

My high-level understanding/hope is that we might get upper picture without any special handling in existing code, by only changing outer perimeter line thickness.

Indeed it is somewhat similar to "alternate extra wall", and might have somewhat similar positive impact on infill adhesion as well.

GregValiant, 2025-01-26:
One thing to always remember is that "Everything affects everything.".  Even small changes can have cascading effects on existing code.  I don't see this as a "small change" but I do think the idea has value.

smartavionics, 2025-01-26:
Hi, it's an interesting idea and probably not too hard to implement. I will look into it this week.

GregValiant, 2025-01-26:
I printed a test piece and visually it looks the same as any other print.
I don't have any way to perform a tensile test.  That would be the ticket.  I did try to break the test piece but that calibration shape of mine is a poor design for testing the layer adhesion.

LilBub, 2025-01-27:
That's a very interesting idea, anything that helps the performance of part strength on the z axis is a win

smartavionics, 2025-01-27:
I have an implementation. I added a setting for the outer wall line width used on even layers. Will test it a bit and make a release before too long so you can play with it. One thought, it may be worth reducing the print speed in proportion to the increase in line width so that the extrusion rate for the outer walls remains constant. I'll have a play with that and possibly add another setting to enable it if it looks worthwhile.

https://github.com/user-attachments/assets/99ec2835-a4e1-4079-b7a6-0023a1de4c10

GregValiant, 2025-01-27:
@smartavionics 
Going from a .4 line width to a .6 will send the flow to 150% so that might well be a consideration.  My test print was at a low speed so the extra flow didn't strain anything.

I suppose you could calculate the Volumetric speed at the normal line width, recalculate at the Even Layer Outer Wall Line Width, and adjust the speed to maintain that flow rate.  Doing that would eliminate the setting but people would wonder (and complain) about their prints slowing down for the outer walls every-other layer.

Maybe a "only adjust the speed if it's over X" might be in order.  My print would have gone from 50 to 33 for the outer walls to maintain the 4.0mm³sec flow rate.  The machine can handle up to 12mm³/sec so I would not have needed/wanted a speed adjustment.
If I had been printing at 100 then yes, it should slow down.

If you added a check box to turn "Outer wall speed adjustment" on or off it would keep the user from having to know their "Max Volumetric Flow Rate" of their machine.  Not a lot of people do.

BarsMonster, 2025-01-27:
@smartavionics Result looks very good! 

Odd and even perimeter speeds can be the same by default, derived from perimeter speed, but can be editable. 

Many people print outer perimeters substantially slower than their max volumetric flow, so indeed they might choose not to adjust it. But for fast prints separate adjustment of perimeter speeds indeed might be needed.

smartavionics, 2025-01-28:
I have just made a new release that embodies this feature. You can find it at https://github.com/smartavionics/Cura/releases.

If you have any feedback/issues when using it, please open an issue at https://github.com/smartavionics/Cura/issues rather than spamming the UM repo.

Thanks.

BarsMonster, 2025-01-28:
@smartavionics I've tested it - and it seems to be perfect both for simple and complex model slicing. Thanks alot!
Flow equalization checkbox also works correctly.

BarsMonster, 2025-02-14:
Regarding patent US11331848B2 :
https://patents.google.com/patent/US11331848B2/en?oq=16%2f910%2c556

My understanding is that my proposal is not covered by 2 claims in the patent text. 

Patent is covering depositing "about one-half of the width", which is impractical for 3d printing (nozzle cannot reliably extrude less than 100% of it's width). In my proposed example line thickness variation is from 100% to 150% (or 66% to 100% if you re-normalize). This variation in default configuration can be even slightly decreased, to 100/133% (75%/100%) to make it as far as mathematically possible from 50/100%. Fortunately we don't even need to have default configuration here. 

Also, it is suggested we allow thickness variability on the inner side to enhance adhesion in infill, which is missing in the patent.

Finally, for maximum legal safety, we should avoid introducing "Simplified brick layers' as a checkbox, but rather have 2 separate outer wall thicknesses, equal by default, End user makes his choice on variability.

BarsMonster, 2025-02-14:
Finally, to protect subsequent ideas from patenting, here is public disclosure of further improvements:

1) One of the disadvantages of original idea in this feature request is potential loss of finest details due to some outer perimeters being thicker. This can be resolving by introducing variability to the second perimeter (counting from the outside), not the first. This will slightly reduce layer adhesion strength, but surface details would be as good as it can be. 

Also, this further increases legal distance from the prior art idea in patent US11331848B2. 

2) To further improve adhesion to infill it is possible to increase stick-out distance into infill more than 50% of nozzle width (for half of layers).

For example, if we are printing with 0.4mm nozzle, 0.2mm layer height - stick-out distance into infill in the original proposal is 50% of 0.4mm = 0.2mm, with 0.4mm (100%) being in the perimeter. Maximum adhesion strength might be achieved if stick-out is approximately 0.3mm / 75% of nozzle width, with 0.3mm / 75% of nozzle width is part of the perimeter (i.e. half of thickened line width sticks out into infill). 

To achieve that line thickness must be gradually increased when going from outer perimeter to inner one.

3) Ideas 1 and 2 can be combined:
Outer perimeter have unchanged line width, but for odd or even layers - perimeter line width increases when printing perimeters from outer one to inner one so that inner-most perimeter sticks out into infill by 50% of thickened line width (i.e. for 0.4mm nozzle, 0.6mm thickened line width - infill stickout is 0.3mm).

This will have maximum improvement of infill adhesion, likely even more than when using "alternate extra wall", without consuming additional print time (if we are not limited by max flow rate). Can slightly increase perimeter adhesion, but not as much as in other variants. Finally, this again further separates us from patented ideas. 

So apparently there is quite alot of fun stuff we can do by varying line thicknesses :-)

bondarchook, 2025-06-02:
New insights into the "brick layers" technique from CNC Kitchen:
https://www.youtube.com/watch?v=dDgA51zdfLc

TLDR: need to increase flow while using the "brick layers" technique for maximum strength.