Ultimaker/Cura #20190: Portion of object defined in .stl file fails to slice so only part of the object prints from the .gcode file generated

3DPrinting120, 2025-01-27:
### Cura Version

5.9.0 (Also 4.5)

### Operating System

Windows 11 Home

### Printer

Creality Ender 3

### Reproduction steps

1. Created .stl using Autodesk Fusion (a cylinder with a 2 mm depth, narrow cylindrical cutout adjacent to the OD surface)
2. the .stl loaded in CURA properly but the OD wall adjacent the cutout was missing in the sliced preview 
3. Uninstalled CURA 5.9.0, downloaded and reinstalled - tried again - same issue
4. Tried CURA 4.5 - same issue
5. Tried CURA 3.6 - sliced successfully!
6. Tried Bambu Studio - sliced successfully!
7. Attaching .stl file, CURA 5.9.0 screenshots of successfully loaded .stl and incomplete preview image after slicing, cura.log and project file in .zip file below

### Actual results

.gcode files from CURA 5.9.0 and also CURA 4.6 failed to print the entire object as defined by the .stl file

### Expected results

Both Bambu Studio V 1.10.1.50 and CURA 3.6 slicers provided successful prints of the complete object depicted in the .stl file

### Add your .zip and screenshots here ⬇️

[CURA Bug.zip](https://github.com/user-attachments/files/18553254/CURA.Bug.zip)

GregValiant, 2025-01-27:
Thanks for the report.
Given the Cura default settings, this is expected behavior.
That outer rim-wall is only 0.3mm thick.  For a 0.40 nozzle, that is below the default "Minimum Line Width" of 0.34 and so it will not slice.
If you were to enable "Print Thin Walls", or if you set the "Minimum Line Width" to 0.30 it will slice.

[image]

This appears to be a setting issue so I'll remove the bug label and close this.  I hope you understand.

closed 2025-01-27 by GregValiant

3DPrinting120, 2025-01-27:
Thank you for your reply - followed your instruction and works just as you said.