Ultimaker/Cura #20195: Unable to slice an STL file (crashes or hangs)

Xeno8888, 2025-01-27:
### Cura Version

5.9.0

### Operating System

macOS 15.3 (24D60)

### Printer

Anycubic i3 Mega

### Reproduction steps

Tried to slice a downloaded STL file from Thingiverse - but slicing either crashes or hangs.

### Actual results

Hang or crash:
```
2025-01-27 18:10:42,224 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.224] [info] Starting export...
2025-01-27 18:10:42,288 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1065]: Backend exited abnormally with return code 1!
2025-01-27 18:10:42,288 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [434]: Attempting to kill the engine process
2025-01-27 18:10:42,289 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [440]: Killing engine process
2025-01-27 18:10:42,289 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [443]: Engine process is killed. Received return code 1
```

### Expected results

A sliced result.

### Add your .zip and screenshots here ⬇️

```2025-01-27 18:10:42,145 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.140] [info] Progress: start accomplished in 0.000s
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.140] [info] Starting slice...
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.140] [info] Slicing model...
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.140] [info] Slice of mesh took 0.000 seconds
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.141] [info] Make polygons took 0.001 seconds
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.141] [info] Progress: slice accomplished in 0.001s
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.141] [info] Starting layerparts...
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.141] [info] Progress: layerparts accomplished in 0.000s
2025-01-27 18:10:42,146 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.141] [info] Starting inset+skin...
2025-01-27 18:10:42,159 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.159] [info] Progress: inset+skin accomplished in 0.018s
2025-01-27 18:10:42,159 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.159] [info] Starting support...
2025-01-27 18:10:42,224 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.224] [info] Progress: support accomplished in 0.065s
2025-01-27 18:10:42,224 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.224] [info] Starting export...
2025-01-27 18:10:42,288 - ERROR - [MainThread] CuraEngineBackend.CuraEngineBackend._onBackendQuit [1065]: Backend exited abnormally with return code 1!
2025-01-27 18:10:42,288 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [434]: Attempting to kill the engine process
2025-01-27 18:10:42,289 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [440]: Killing engine process
2025-01-27 18:10:42,289 - DEBUG - [MainThread] CuraEngineBackend.CuraEngineBackend._terminate [443]: Engine process is killed. Received return code 1
2025-01-27 18:10:42,289 - DEBUG - [MainThread] UM.Backend.Backend._createSocket [297]: Previous socket existed. Closing that first.
2025-01-27 18:10:42,290 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [232]: Socket state changed to Listening
2025-01-27 18:10:42,306 - INFO - [MainThread] UM.Backend.Backend.startEngine [96]: Started engine process: /Applications/UltiMaker Cura.app/Contents/Resources/CuraEngine
2025-01-27 18:10:42,306 - DEBUG - [MainThread] UM.Backend.Backend._backendLog [140]: [Backend] Calling engine with: ['/Applications/UltiMaker Cura.app/Contents/Resources/CuraEngine', 'connect', '127.0.0.1:49675', '']
2025-01-27 18:10:42,315 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.314] [info] Sentry url: https://734f9ec9024f73e53701d59c3ffddfe3@o323038.ingest.sentry.io/4506257745510401
2025-01-27 18:10:42,315 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.315] [info] Sentry config path: /Users/*******/Library/Application Support/cura/.sentry-native
2025-01-27 18:10:42,315 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.315] [info] Starting sentry
2025-01-27 18:10:42,320 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,321 - DEBUG - [MainThread] UM.Backend.Backend._logSocketState [236]: Socket state changed to Connected
2025-01-27 18:10:42,321 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] Cura_SteamEngine version 5.9.0
2025-01-27 18:10:42,321 - DEBUG - [MainThread] UM.Backend.Backend._onSocketStateChanged [225]: Backend connected on port 49675
2025-01-27 18:10:42,321 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,321 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] Copyright (C) 2024 Ultimaker
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] This program is free software: you can redistribute it and/or modify
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] it under the terms of the GNU Affero General Public License as published by
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] the Free Software Foundation, either version 3 of the License, or
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] (at your option) any later version.
2025-01-27 18:10:42,322 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] This program is distributed in the hope that it will be useful,
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] but WITHOUT ANY WARRANTY; without even the implied warranty of
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] MERCHANTABILITY or FITNESS FOR A PARTICULAR PURPOSE.  See the
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] GNU Affero General Public License for more details.
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend]
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] You should have received a copy of the GNU Affero General Public License
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] along with this program.  If not, see <http://www.gnu.org/licenses/>.
2025-01-27 18:10:42,323 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.320] [info] Connecting to 127.0.0.1:49675
2025-01-27 18:10:42,424 - DEBUG - [EngineOutputThread] UM.Backend.Backend._backendLog [140]: [Backend] [2025-01-27 18:10:42.424] [info] Connected to 127.0.0.1:49675```

![Image](https://github.com/user-attachments/assets/0ebe459f-6bd0-4ca4-8fc3-e6120fc6ae4c)

[cura.log.zip](https://github.com/user-attachments/files/18562051/cura.log.zip)

[AI3M_donglefix.3mf.zip](https://github.com/user-attachments/files/18562085/AI3M_donglefix.3mf.zip)

GregValiant, 2025-01-28:
Thanks for the report.
It won't slice with your "Fine 0.2 - hmm slow init la xxx" profile active but when I switch to "Normal" and "Keep changes" it slices as it should.  The problem seems to be something in that profile.  All the other profiles work but that one crashes with:
_"Backend exited abnormally with return code 3221225477!"_
in the log file.  That's a generic error code.

The Cura team will take a look.

Xeno8888, 2025-01-28:
@GregValiant Many thanks for having a look at it - I never had an issue with that profile until now.
I hope the team can find the cause and make Cura even better (or tell me what I did wrong).