Ultimaker/Cura #20197: Multi Color based on Vertex color of STL

ToyboxZach, 2025-01-27:
### Is your feature request related to a problem?

Multi color printers are becoming more and more popular and I think cura should be moving to be able to support them. 

To do a multi color model I see two ways I would want to input into cura.

1. Add multiple distinct objects are different colors and print each object as a different color. My understanding of this is that cura already handles that which can handle a lot of different cases.

2. Be able to "paint" on colors to an existing model. Generally you would want to color a face or vertex to a color, but creating a solid out of it would be a "hard" problem, but it can be specified in the Vertex color of the model.

### Describe the solution you'd like

I think Cura should be able to respect the vertex colors of an stl and be able to separate a model based on the different colors it has. I think there are plenty of cases where just getting the wall colors to be right when specified would be super helpful, and then the cura engine could just decide to do the infill whichever color was convenient. 

Are there any plans to support this style of multi color? Otherwise I think I'm going to start looking at prusa which does support it.

### Describe alternatives you've considered

The alternative is to only allow the per object colors, but that is pretty restrictive on what tools you can use to generate it.

### Affected users and/or printers

Multi color 3d printers is definitely a growing market https://www.tomshardware.com/3d-printing/best-multicolor-3d-printers

Asterchades, 2025-01-28:
There's no standard for colouring vertices in STL files - it's a raw geometry format. Any implementation would either need to be based on an unofficial extension (such as the SolidView version that uses the Attribute property for 15-bit BGR colour) or something unique to Cura, neither of which seems particularly useful as someone will invariably want to use a file made with different rules.

This is one of the many extensions made to the 3MF format over the STL format. My understanding is that Cura can already import multi-object 3MF files but I don't know if it automatically assigns anything based on the colours, instead requiring the user to assign each to the appropriate tool. Perhaps someone more versed in multi-material/-extruder printing can chime in with a additional feedback on this particular functionality.

ToyboxZach, 2025-01-28:
In my view of it, it doesn’t actually matter what the colors themselves mean. As long as you can bucketize the values and assign each “color” to an extruder it would enable you to specify which extruder is which set of “colors”

The core of this idea, is building cura up to able to decide how the color is seperated in the volume based on the color shell, instead of putting that pressure on the 3D design tools