Ultimaker/Cura #20202: Cura configure only opens when I initially open the program.

Randeb74, 2025-01-27:
### Cura Version

5.9.0

### Operating System

Mac OS Sequoia

### Printer

Prusa i3

### Reproduction steps

I can only access the Cura configuration settings on a fresh opening of the program. After I load a file I have to close the program and reopen to access the config settings. Not a big deal but a bit annoying when you're trying to get a new printer setup working right. Thanks.

### Actual results

First comments says it all.

### Expected results

First comment says it all.

### Add your .zip and screenshots here ⬇️

Nothing to add.

GregValiant, 2025-01-28:
Thanks for the report.
That doesn't happen on my Win10 system and there have not been other reports of this behavior.  It might be system-specific.

- Open Cura and recreate the condition.
- Use the "Help | Export package for technical support" command then zip that file and post it here (not via email).

Maybe the log file will give a clue as to what is going on.

Randeb74, 2025-01-28:
[Compress.zip](https://github.com/user-attachments/files/18574316/Compress.zip)

Hi Greg, here is the file that you requested. Thank you for taking a look at this issue.

GregValiant, 2025-01-28:
The Cura team is busy getting ready for a new release so it may be a while before they get to this.
I don't see anything obvious in the log file so you will need them to take a look.

Randeb74, 2025-01-28:
Thank you, that’s fine with me. Great service snd a great free program regardless! Randy