Ultimaker/Cura #20207: Side Bar Menu (support block Icon missing)

rcquad808, 2025-01-28:
### Cura Version

5.9.0

### Operating System

Windows 10 Pro

### Printer

Sovol SV01

### Reproduction steps

The support Icon in the side bar menu is not there anymore. I can't find anywhere. Is there a plugin I need to install? Am I missing something?

[image]

[image]

### Actual results

I can't seem to find it in Cura.

### Expected results

My results are fruitless.

### Add your .zip and screenshots here ⬇️

No comments

GregValiant, 2025-01-28:
Thanks for the report.
Your first image is an older version of Cura.  The icons have changed.
Your bottom image looks like current versions and the support blocker button is still the sixth one down.

[image]

rcquad808, 2025-01-28:
Greg,

I clicked on the icon and nothing happen. Before it would bring up the different types of supports to chose form but, now nothing comes up. This is what I get when I click on it. 

[image]

GregValiant, 2025-01-29:
There was a "Custom Supports" plugin but it was removed from the MarketPlace.  What I indicated is the regular "Support Blocker" tool.  They can be scaled and configured to do different things, but they are always rectangular solid shaped.

I looked around.  It appears that Krasimir Stefanov moved his Github page.
You can try download that version of the plugin from [HERE](https://github.com/lokster/cura-custom-supports/releases/tag/v0.1.0).  It appears that it works with Cura 4.x but not with the latest versions of Cura (the textboxes and other controls changed in 5.0).

Maybe he just didn't feel like updating the plugin to the latest QT controls used by Cura.

rcquad808, 2025-01-29:
Okay thank you....

closed 2025-01-29 by rcquad808

fieldOfView, 2025-01-29:
There's also this plugin:
https://github.com/5axes/CustomSupportCylinder

You will have to install it by downloading a curapackage file from the releases page, and dropping that file onto the buildplate as if you are opening a 3d model.
https://github.com/5axes/CustomSupportCylinder/releases/tag/V2.8.0

rcquad808, 2025-01-29:
Yes, that's is the one I was looking for. It's working now. Thank you so much!