Ultimaker/Cura #20212: Slicing repeatedly fails on Cura 5.9 for common open-source model

drollicdotcom, 2025-01-29:
### Cura Version

5.9

### Operating System

Windows 11

### Printer

Elegoo 3 Plus

### Name abnormal settings

Tree supports were enabled

### Describe model location

Multiple orientations tried, all failed

### Describe your model

Simple model commonly used with popular multiboard organizer

### Add your .zip here ⬇️

[EN3P_20x7.5 mm Hook (Bolt-Locked Insert).zip](https://github.com/user-attachments/files/18592835/EN3P_20x7.5.mm.Hook.Bolt-Locked.Insert.zip)

drollicdotcom, 2025-01-29:
Was downloaded from [this page](https://thangs.com/designer/Multiboard/3d-model/20x7%C2%B75%20mm%20Hook%20%28Bolt-Locked%20Insert%29-1131710).

HellAholic, 2025-01-29:
Model slices without issue in the project file you included.

[image]

[image]

[image]

Tried repairing the model, give that version a try if it doesn't work go under Help > Export Package for technical support and share that zip file, that package includes the Cura logs which might give a clue as to why you're experiencing the slice fail.

[EN3P_20x7.5 mm Hook (Bolt-Locked Insert)-Repaired.zip](https://github.com/user-attachments/files/18594779/EN3P_20x7.5.mm.Hook.Bolt-Locked.Insert.-Repaired.zip)

closed 2025-02-13 by github-actions[bot]