Ultimaker/Cura #20213: Import Filter Problem V5.8 works, but not 5.9.

xarminius, 2025-01-29:
### Cura Version

5.9.0

### Operating System

Win 10 

### Printer

Not of importance

### Reproduction steps

I've created an STL with freecad. It's a flat ring and something is put on the top. 
(I append a zip). 
5.8 produces a fine  model in gcode, V5.9 produces scrap (It produces a round baseplate, BUT it's a ring) 
I checked out in "meshlab". My grid seems to be OK. 

[hdyhltr_Obt2_v05-Krpr.zip](https://github.com/user-attachments/files/18593373/hdyhltr_Obt2_v05-Krpr.zip)

kind regards
Hermann

### Actual results

The model what you see (preview) is wrong.

### Expected results

Please check out the import filter. 

### Add your .zip and screenshots here ⬇️

[image]

[image]

HellAholic, 2025-01-29:
In printer settings uncheck the "Remove all holes"

With the setting enabled in 5.9
[image]

Without the setting
[image]

closed 2025-01-29 by HellAholic

xarminius, 2025-01-30:
Hello HellAholic!
Thank you for Info. But this part has 2 open holes when you check out and look at the front.
They stay open. I am confused.
Nevertheless- thank you.