Ultimaker/Cura #20214: Build plate displays in front of printer model and

fhoran, 2025-01-30:
### Cura Version

5.9

### Operating System

macOS So14.6.1 (23G93)

### Printer

Ultimaker S5 as well as custom printer profile

### Reproduction steps

Load in a model on any pinter that has its frame/gantry stored in share>cura>resources>meshes

### Actual results

The mesh for the 3D printer displays behind the print bed but in front of any models that you load in, this is an issue with SCARA configuration machines where it is helpful to see intersections between the mast and build volume, not just disallowed areas.
[image]
[image]
[image]

### Expected results

The mesh of the printer should display transparent on top of the print bed where it intersects. These pictures are from Cura 4.13.1 where it displays properly.

[image]
[image]
[image]

### Add your .zip and screenshots here ⬇️

[3DPSV3_DemoCube.3mf.zip](https://github.com/user-attachments/files/18595997/3DPSV3_DemoCube.3mf.zip)

GregValiant, 2025-01-30:
Thanks for the report.
This appears to be a printer definition error rather than a bug so I will remove the label and close this.  Let me know if you feel that is premature.

When I install a standard UM S5 the platform loads as it should.
[image]

The build plate meshes are models that are handled differently.  Their location in space can be set within the Metadata of a definition file using "Platform_offset".  (I don't recall seeing a "rotation" employed.)

This is from the S5 definition file
Line 11:
`       "platform": "ultimaker_s5_platform.obj",`

Line 42:

        "platform_offset": [
            0,
            -30,
            -10
        ],
        "platform_texture": "UltimakerS5backplate.png",

I have too much time on my hands so I modeled my entire printer as my "platform".  It took some setting up to get the build plate midpoint to 0,0,0.  I did not have to use any "offsets" for this.
[image]

You need to define offsets in your custom definition file, or open the platform in another app and move the platform so the midpoint of the top surface is at 0,0,0.

closed 2025-01-30 by GregValiant

fhoran, 2025-02-08:
Thank you for the thorough response! 

Like you I also have too much time on my hands which is why I modeled my printer in its entirety as well. The issue I was having is that it used to display a translucent version of the printer model **over** the build plate (second 3 images) but in 5.9 the rendering order is incorrect (first 3 images). This printer is a SCARA so the printer model intersects with the build volume (thats why there are the dark gray disallowed areas around the printer model)

It used to display in order of what is closest to the camera. Was it a stylistic decision to always have the print bed display in front? If so it is still a bug because the printer model displays in front of any object being sliced but behind the build plate (3rd image from before)

I can reproduce the issue with an S5 but it is less overlap so it is less noticeable so I highlighted the issue in red.

4.13.1 - Translucent printer model occludes print bed on S5
[image]
5.9 - Transluscent printer model does not occlude print bed (red marks where it is supposed to show) on S5

[image]

[image]

GregValiant, 2025-02-09:
Post the "platform" file here and I'll take a look.  The images are very odd.
Did you use a "Custom FFF" printer in Cura, or did you have a printer specific definition file?

fhoran, 2025-02-15:
I have a printer specific definition file, I made it for a commercial printer that didn't have cura support. It is a SCARA machine from 3d Potter. I have a stl in resources>meshes, I do not have a png image in resources>images.

Do you want the platform file from the S5 or my personal printer? Since the display issue happens with both.

When you rotate the view of your personal printer 90 degrees does the gantry show up on top of the build plate or is it hidden behind it?

GregValiant, 2025-02-15:
My printer platform file is an STL.  Cura does the coloration.  I have no idea how.
I would think that the file itself must be whole and complete with no errors in the mesh.
When I first tried to get the custom platform into Cura, it was difficult to position it correctly in space.  I used MS 3 builder and moved the model, check it in Cura, move the model some more.  Eventually, I got the positioning spot on.
There are a lot of printer definitions that include complex platforms.  Just checking yours for errors might reveal something that is confusing Cura.  They are models like any other and need to be correct in all respects.  No extra faces, water-tight...that sort of thing.

fhoran, 2025-02-24:
It seems I am not communicating well, the printer mesh is closed and positioned correctly but when the scene is rendered, it is not rendering based on distance to the camera. Up until 5.9 & now 5.9.1 whatever was closest to the camera was displayed first.

This issue effects all printers that I have checked. 
With 5.9.1 and this still being an issue should I restate the problem only using ultimaker printers in a new bug report, or is there a way for me to explain what I mean in a video call.

I am worried it was a stylistic change but that still doesn't explain the out of order rendering I mentioned earlier.

GregValiant, 2025-02-24:
This is with an S5 active.
I think this is what you mean in regards to that printer.  The rear wall is incomplete where it crosses the build plate?
[image]

This is with my printer active.  The camera shot is from the rear and is looking through the right Z upright and the power supply.  Those are invisible where they cross the build plate.

[image]

This is a similar image from the rear but I've stuck a model on the built plate.  Where the upright and power supply occlude the model, the model is obviously behind them and the part that is not behind the platform STL shows full color.

[image]

I'll re-open this.  (I'm not sure what anyone will do with it or what sort of priority it will get at UM).