Ultimaker/Cura #20215: [5.9.0] Software will not start

kellockc, 2025-01-30:
### Cura Version

5.9.0

### Operating System

Windows 11 Pro

### Printer

Creality Ender 3 Pro

### Reproduction steps

Any time I try to start any version newer than 5.0.0, it does not load.  5.0.0 works fine on this PC.  Have tested running both Windows 10 Pro and now upgraded to Windows 11 Pro.  I get the spinning cursor, but then it goes away and the software never loads.  When I try to open Task Manager, it does not show Cura as even running to shut it down.  

### Actual results

Software never executes, even when run as administrator, even though I am logged in as the administrator.

### Expected results

I expect software to load when it is started, duh...

### Add your .zip and screenshots here ⬇️

There is nothing to add, nothing to show, as the software simply does NOTHING!

Asterchades, 2025-01-30:
A log file should have been created which might offer some insight into what's happening. It will be exactly where it was described when creating this issue.

> - 🔵 Unexpected crashes and behavior
> Please add a log file with information on what your Cura is doing.
> You can find your log file here:
> Windows: %APPDATA%\cura\<Cura version>\cura.log
> MacOS: $USER/Library/Application Support/cura/<Cura version>/cura.log
> Ubuntu/Linux: $USER/.local/share/cura/<Cura version>/cura.log

(eg Cura 5.9 under Windows would be _%appdata%\cura\5.9\cura.log_)

In the unlikely event that no log file was created, you're probably encountering a system-level issue. In which case the Windows *Reliability Monitor* is going to have either a warning or an error associated with the failed launch attempts. The details here may help to figure out what's going on, although often they're of limited use to non-Microsoft technicians.

kellockc, 2025-01-30:
There is no log file.  I did check.

I did find the following in the reliability tool...
-----------------------------------------
Source
UltiMaker-Cura.exe

Summary
Stopped working

Date
‎1/‎29/‎2025 12:21 PM

Status
Report sent

Description
Faulting Application Path:	C:\Program Files\UltiMaker Cura 5.9.0\UltiMaker-Cura.exe

Problem signature
Problem Event Name:	BEX64
Application Name:	UltiMaker-Cura.exe
Application Version:	0.0.0.0
Application Timestamp:	673b75a8
Fault Module Name:	ntdll.dll
Fault Module Version:	10.0.22621.4541
Fault Module Timestamp:	e7035eba
Exception Offset:	00000000000a53d0
Exception Code:	c0000409
Exception Data:	000000000000000a
OS Version:	10.0.22631.2.0.0.256.48
Locale ID:	1033
Additional Information 1:	bec0
Additional Information 2:	bec0472d7b6cf52f876ebc166d8fbf91
Additional Information 3:	7938
Additional Information 4:	793859376a275c3bdc96698a84657124

Extra information about the problem
Bucket ID:	1f31df3b3d364d1d82216fe9a20387a1 (1306448416150357921)

Asterchades, 2025-01-30:
BEX64 (Buffer EXception) is a DEP (Data Execution Protection) error within a 64-bit Windows environment. Basically it means that something tried to access memory that it shouldn't be accessing.

I do know that this can occur as a result of Cura interacting with multiple displays incorrectly. You can correct this by editing the _cura.cfg_ file (found in the same directory as the log should have been) and removing the lines that start with *window_* from the [general] section - there should be up to five such lines in all. Any text editor, including NotePad, is sufficient for this. Should this fix it, you can prevent this occurring again by unchecking the _Restore window position on start_ setting in the Cura preferences.

I have also seen it occur on multi-GPU systems (such as those with a dGPU and iGPU) where a surface is obtained for one GPU while the program attempts to launch on another. In this case you need to use your GPU's control panel to force Cura onto a specific GPU, rather than leaving it on auto. This primarily affects laptops as generally iGPUs are disabled on desktop machines if they're not connected to a display.

Should neither of those be the issue, that's when things get complicated. At that point you need to start investigating other programs you've got installed which might be interacting with Cura, as Cura on its own shouldn't be causing this (there would be a lot more complaints if it did). Programs that have overlays or GPU-monitoring features (EVGA's PrecisionX, MSI's Afterburner, RTSS, etc) or hook into controls (Razer's Synapse, Logitech's LGS, Corsair's iCUE, etc) when applications are launched are prime candidates, but they aren't the only possibility.

kellockc, 2025-01-30:
Edited the 3 lines out, did not help.  Disconnected the second monitor, rebooted, did not help.  I also made sure the Intel graphics tray tool was disabled.

Yet 5.0.0 works flawlessly on this machine...

Almost nothing else is installed on this machine.

Asterchades, 2025-01-30:
Disabling the tray tool is not the same as disabling the GPU entirely or setting an application to use a specific GPU. It's a bit of a "hail Mary" anyway, but details on how to specify which GPU a program is to use can be found here: https://support.makerbot.com/s/article/1667412437627

While reaching, it's also known that some versions of the Nahimic audio service used by Steelseries headsets can cause problems. Though it also caused problems with 5.0, so the chances of it being at fault here feel slim. Still, it's worth mentioning just in case a fix was deployed for earlier Cura versions that no longer works with more recent ones.

Unfortunately further diagnostics on this issue is beyond my capabilities. I'll have to defer this to someone further up the chain. If I can get @HellAholic's attention hopefully they or someone else more capable than me can chime in.

kellockc, 2025-01-30:
Pretty sure this PC only has one GPU, being an all-in-one, not a laptop.  Only video adapter that shows in the specs is Intel HD Graphics 630.  And being an all-in-one, I can't just swap it for a new card...  I just can't get why 5.0.0 is working flawlessly but keeps telling me there are new versions, and the new versions refuse to execute...

HellAholic, 2025-01-30:
Aloha, attention on deck.
So these crashes could have a range of different reasons, and unfortunately it's not a one shot solution. But I do see that you're pretty savvy so makes the issue handling a bit easier but still tedious for you. 
I will just list all the possible things that come to my head and please go through 1 by 1 and check to see which one resolves it for you and if not let me know.

1. Check for DEP Conflicts

    Open System Properties (Win + R → sysdm.cpl → Advanced → Performance → Settings).
    Go to Data Execution Prevention (DEP) tab.
    If Cura is listed under Turn on DEP for all programs except those I select, try removing it.
    If it’s not listed, try adding Cura manually and test.

2. Run Cura as Administrator [ you already tested so gonna jump over  but as a reference for someone else that finds this report later]

    Right-click UltiMaker-Cura.exe → Run as Administrator.

3. Check GPU Drivers

    Outdated or incompatible GPU drivers can cause crashes.
    Update your drivers.
    [Another user had a BSOD event when using cura and that one was tied to drivers of motherboard / graphics, it's worth a try]

4. Disable Compatibility Mode

    Right-click UltiMaker-Cura.exe → Properties → Compatibility.
    Ensure Run this program in compatibility mode is unchecked.
    If it's unchecked move to the next one.

5. Check for Corrupt Files

    Reinstall Cura: Uninstall Cura and reinstall it.
    If the issue persists, check C:\Users\<YourUser>\AppData\Roaming\cura\5.9 for crash logs.

6. Check for Plugins & issue with upgrade

    If you installed any Cura plugins, one might be causing the crash.
    Alternatively, there were some crashes / bad behaviors caused by the corrupted data
    Easiest solution for this is to keep a backup of your `user/Local/cura` & `user/Roaming/cura` and after the back up delete the cura folders in both locations.
    

7. Disable Overclocking [ Do not think you have overclocking on a all-in-one but reference for others]

    If your system is overclocked, reset to stock settings.

8. Event Viewer Logs

    Open Event Viewer (Win + R → eventvwr.msc).
    Navigate to Windows Logs → Application and check for related Cura crash logs.

Try these steps and let me know if any of them helped.

kellockc, 2025-01-30:
Thanks again for the assist.
1.  Was set for DEP for Windows only, tried changing and adding Cura, no change.
2. Already tested, no luck
3. Checked for updated drivers from intel, testing now
4. Wasn't on
5. Uninstalled and reinstalled into a directory off the C: drive not in Program Files, which is how I have 5.0.0. installed.  No change.
6. Never had a chance to install anything, it won't open, and I have no extensions in 5.0.0
7. Not overclocked
8. Event Viewer does show an error.  Will paste below.

Log Name:      Application
Source:        Application Error
Date:          1/30/2025 12:49:53 PM
Event ID:      1000
Task Category: Application Crashing Events
Level:         Error
Keywords:      
User:          LABPC\Admin
Computer:      LabPC
Description:
Faulting application name: UltiMaker-Cura.exe, version: 0.0.0.0, time stamp: 0x673b75a8
Faulting module name: ntdll.dll, version: 10.0.22621.4541, time stamp: 0xe7035eba
Exception code: 0xc0000409
Fault offset: 0x00000000000a53d0
Faulting process id: 0x0x2DF4
Faulting application start time: 0x0x1DB733F5D817695
Faulting application path: C:\UltiMaker Cura 5.9.0\UltiMaker-Cura.exe
Faulting module path: C:\WINDOWS\SYSTEM32\ntdll.dll
Report Id: d5e0d190-63c1-4aef-b02d-2909354ffe1c
Faulting package full name: 
Faulting package-relative application ID: 
Event Xml:
<Event xmlns="http://schemas.microsoft.com/win/2004/08/events/event">
  <System>
    <Provider Name="Application Error" Guid="{a0e9b465-b939-57d7-b27d-95d8e925ff57}" />
    <EventID>1000</EventID>
    <Version>0</Version>
    <Level>2</Level>
    <Task>100</Task>
    <Opcode>0</Opcode>
    <Keywords>0x8000000000000000</Keywords>
    <TimeCreated SystemTime="2025-01-30T17:49:53.9863144Z" />
    <EventRecordID>3566</EventRecordID>
    <Correlation />
    <Execution ProcessID="12136" ThreadID="2344" />
    <Channel>Application</Channel>
    <Computer>LabPC</Computer>
    <Security UserID="S-1-5-21-2476890436-3311477373-3142938928-1001" />
  </System>
  <EventData>
    <Data Name="AppName">UltiMaker-Cura.exe</Data>
    <Data Name="AppVersion">0.0.0.0</Data>
    <Data Name="AppTimeStamp">673b75a8</Data>
    <Data Name="ModuleName">ntdll.dll</Data>
    <Data Name="ModuleVersion">10.0.22621.4541</Data>
    <Data Name="ModuleTimeStamp">e7035eba</Data>
    <Data Name="ExceptionCode">c0000409</Data>
    <Data Name="FaultingOffset">00000000000a53d0</Data>
    <Data Name="ProcessId">0x2df4</Data>
    <Data Name="ProcessCreationTime">0x1db733f5d817695</Data>
    <Data Name="AppPath">C:\UltiMaker Cura 5.9.0\UltiMaker-Cura.exe</Data>
    <Data Name="ModulePath">C:\WINDOWS\SYSTEM32\ntdll.dll</Data>
    <Data Name="IntegratorReportId">d5e0d190-63c1-4aef-b02d-2909354ffe1c</Data>
    <Data Name="PackageFullName">
    </Data>
    <Data Name="PackageRelativeAppId">
    </Data>
  </EventData>
</Event>

kellockc, 2025-01-30:
Lenovo did not show any updated drivers for this machine, nor did Windows Update, so I went directly to Intel and downloaded a new driver.  And it worked!  5.9.0 is now up and running.  Thanks everyone for the suggestions!

Asterchades, 2025-01-31:
Given it now appears to be working I'm going to close this up. Cheers to @HellAholic for stepping in with the appropriate solution - I never would have suspected GPU drivers as OpenGL is rarely touched (though I guess there's still plenty of Windows back-end stuff that can be altered).

closed 2025-01-31 by Asterchades