Ultimaker/Cura #20217: G-Code Script for Printing of Temp-Tower

joker-mik, 2025-01-30:
### Is your feature request related to a problem?

Hello,
if you print temperature towers frequently, then it is annoying to set up many individual scripts (ChangeAtZ). 

### Describe the solution you'd like

It would be nice to have a script for this task. In the following project there is a script that is supposed to do exactly that - unfortunately it no longer works.
https://www.thingiverse.com/thing:2862812
It would be nice if such a script were implemented in Cura by default.

### Describe alternatives you've considered

 The alternative is to set up many individual scripts (ChangeAtZ). However, this is prone to errors.

### Affected users and/or printers

This benefits all users who print temperature towers.

Asterchades, 2025-01-30:
There are two existing plugins on the Marketplace which already have this sort of functionality: AutoTowers Generator and Calibration Shapes. AutoTowers is more self-contained, in that using it to add the assorted towers (which are not limited only to temperature) will automatically apply an appropriate script depending upon the tower model being added. Calibration Shapes, however, adds a series of post-processing scripts which can be activated separately (and need to be to use the towers at all) that can be reconfigured should your model of choice not match the included models.

GregValiant, 2025-01-31:
@joker-mik, let us know if Asterchades suggestions work for you.

I understand that using ChangeAtZ was clumsy.  Strictly speaking, there are no "calibration tower" scripts that are packaged with Cura, you do have other options in the Marketplace for altering the gcode so it will work with the various calibration towers.

joker-mik, 2025-01-31:
Hello @Asterchades and @GregValiant ,
Thank you very much for your quick replies. The suggestions look very promising.

Shame on me, I didn't even know about the marketplace.

From my point of view, the ticket can be closed. 

Thank you very much.

GregValiant, 2025-01-31:
Done.

closed 2025-01-31 by GregValiant