Ultimaker/Cura #20224: fan control also adaptable for overhangs

lwbsbde, 2025-01-31:
### Is your feature request related to a problem?

fan control also adaptable for overhangs

### Describe the solution you'd like

fan control also adaptable for overhangs

### Describe alternatives you've considered

fan control also adaptable for overhangs

### Affected users and/or printers

All user

GregValiant, 2025-02-01:
The post processor "Advanced Cooling Fan Control" can turn the fan on and off according to your layer selection.  You can turn it on just for those overhang layers.
There is also a "By Feature" option.  Because it's a post-processor and because there is nothing in a gcode file to indicate where an overhang occurs, there is no setting specifically for overhangs.

lwbsbde, 2025-02-01:
The fan should only rotate faster on the overhang, and not the entire layer. The slicer recognizes transitions on the part when the supports are placed. Why not also for the fan control?