Ultimaker/Cura #20227: Dual extruder - materials are inverted

phil123456, 2025-01-31:
### Cura Version

5.9

### Operating System

Win 10

### Printer

Flashforge creator pro

### Reproduction steps

printing 2 materials, they get inverted

I red some people fixed it by adding brim

- added brim
- set left extruder as active

problem solved, right ? well no

now that I recreate the project (previous was corrupted somehow), I redo the exact same thing, wrong material is printed

### Actual results

left  extruder : asa
right extruder : hips

printer prints the brim first, being asa instead of hips

### Expected results

I expect printer to print the brim first with the right extruder, not the left one

I keep trying and trying the same steps I did before I just cannot make it work again, THE WRONG MATERIAL GETS PRINTED

### Add your .zip and screenshots here ⬇️

[Flash Dual.zip](https://github.com/user-attachments/files/18625416/Flash.Dual.zip)

GregValiant, 2025-02-01:
Thanks for the report.
One thing to remember is that UltiMaker is in the business of making and selling 3d printers.  They do the heavy lifting on Cura.  Since they don't make an IDEX model there is no reason to pay their people to provide support for them within Cura.  You might be able to come up with workarounds that get your printer to function properly with a Cura gcode.
  
Flash Forge has not submitted any definition files for their machines.  There is a plugin in the MarketPlace that was developed by an independent contributor and does provide some printer definitions.  The "Creator" models are not among them.  That is all the support there is.

I'm going to remove the bug label and close this as your printer simply isn't supported by UltiMaker Cura.

There are a couple of things you can try.
When you print with a single extruder only, make sure you manually set the "Skirt/Brim Extruder" to that extruder, and manually set the "Support Extruder" to that extruder, and finally disable the unused extruder.  That should keep Cura from using the wrong one for anything.
Take a look at how Sovol handled the problem with their SV04.  They have 5 separate printer definitions:  Duplicate, Mirror, Dual, Single 01, and Single 02.  You pick the printer depending on the mode you want to use.  It's clumsy, but it seems to work.

Beyond that I don't know what to say.   I spent some time working with a poster on the UM Forum trying to get the Creator 2Pro and Pro3 to work with UM Cura.  That included writing a custom post-processor that did things like replacing all the G0 commands with G1 commands and re-ordering the parameters of the tool changes and replacing the fan lines.  It still didn't work.

My overall impression was that Flash Forge really wants you to use Flash Print.

closed 2025-02-01 by GregValiant

phil123456, 2025-02-01:
Flashforge Creator Pro is no IDEX (... Creator Pro 2 is)

As a workaround, I went to the settings and selected the left extruder for the brim
in other words, if there is no brim, it is purely up to what the software will "guess"

also if it guess, why does it have to be random ? as a software engineer, I gotta say this is **really suspicious**

also the "heavy lifting on Ultimaker printers only" seems like a weird argument as THE WHOLE WORLD uses Cura

GregValiant, 2025-02-01:
Any "unsupported printer" can be made to work with enough effort.  I'll put the bug label back on this and re-open it.

There is no guessing or "randomizing".  Cura takes the settings and uses them to provide a gcode toolpath.
For a dual extruder printer, it functions as would be appropriate for an UltiMaker dual extruder printer.  For those of us that do not own UM printers (mine is a Creality Ender 3 Pro) if that method isn't appropriate, we need to come up with workarounds.

If both extruders are being used in a print, then both extruders will be used for the skirt/brim.  It is possible that both extruders would also be used for a raft.
There are numerous extruder functions that must be defined.  By default, UM Printers are configured to use T1 (extruder 2) for supports.  If you don't want that to happen, the Support Extruders would need to be explicitly set to T0 (extruder 1).  Same with the Skirt/Brim.  They must be explicitly set.
"wall_extruder_nr"
"wall_0_extruder_nr"
"wall_x_extruder_nr"
"roofing_extruder_nr"
"top_bottom_extruder_nr"
"infill_extruder_nr"
"support_extruder_nr"
"support_infill_extruder_nr"
"support_extruder_nr_layer_0"
"support_interface_extruder_nr"
"support_roof_extruder_nr"
"support_bottom_extruder_nr"
"skirt_brim_extruder_nr"
"adhesion_extruder_nr"        
"raft_base_extruder_nr"
"raft_interface_extruder_nr"
"raft_surface_extruder_nr"

At the top of the "fff" file you posted is this line:
<profile name="Flash Dual" version="**2015-08-01** 21:29:56" **app="S3D-Software 3.0.1"**>  It is 9 1/2 years old and was for a different piece of software.
Did you install your printer in Cura as a "Custom FFF" printer?  The Creator 2 is not available as one of the printers provided with the Flash Forge plugin.

[image]

GregValiant, 2025-02-01:
Doing some searching I came across this site on [GITHUB](https://github.com/eugr/Flashforge-for-Cura/tree/master).  There are definition files, extruder files, startup and ending gcode files.  If you haven't viewed it before it might be worth a look.