Ultimaker/Cura #20231: Voronoi Slicing Issue

shuffleznl, 2025-02-01:
### Cura Version

5.9.0

### Operating System

WIndows 11

### Printer

V-Core

### Describe model location

Model does slice near off-set, however only in one far corner of bed (-60,-60)

### Describe your model

Voronoi lamp shade, not water tight by design, MeshTools cannot fix

### Add your .zip here ⬇️

https://www.printables.com/model/731495-voronoi-table-lamp/files

[V300_Voronoi_Lamp_E27_Base.zip](https://github.com/user-attachments/files/18630582/V300_Voronoi_Lamp_E27_Base.zip)

GregValiant, 2025-02-02:
Thanks for the report.
I can duplicate the problem with your model.
When Cura complains that a model is "not watertight" it means there are open seams between intersections of edges.  I used MS 3D Builder to repair the model and the repaired model sliced.
[image]

This report on the original model is from formware.co/OnLineSTLRepair
-> Analyzing file
--> 0 Naked edges  (?)
--> 0 Planar holes  (?)
--> 0 Non-Planar holes  (?)
**--> 7 Non-Manifold edges  (?)
--> 5 Inverted normals  (?)
--> 8 Duplicate faces  (?)**
--> 0 Degenerate faces  (?)
--> 0 Disjoint shells  (?)
-> Repairing: 100.00%
-> Converting back
-> File repaired
Time needed for repair (ms): 1073
**--> Vertex count changed from 4253 to 5234 (+981)
--> Triangle count changed from 9180 to 11136 (+1956)**

Here is the repaired model.  See if it works for you.

[GV Voronoi_Lamp_E27_Base.zip](https://github.com/user-attachments/files/18630865/GV.Voronoi_Lamp_E27_Base.zip)

shuffleznl, 2025-02-02:
Wow, that was incredibly fast and very helpful response, thanks @GregValiant .
The updated model sliced and is printing!

Assumed the Mesh Tools extension would take care of fixing meshes, will explore other STL/Mesh repair tools.
3D Builder doesnt appear to be available any more (at least not my region EN-UK).
[image]

closed 2025-02-02 by shuffleznl