Ultimaker/Cura #20241: failed to slice

Bacteria26, 2025-02-05:
### Cura Version

4.13.10

### Operating System

windows 11

### Printer

Lulzbot mini

### Reproduction steps

I downloaded the following file; [https://www.printables.com/model/1135717-golf-tees-6mm-60mm/files](url) the 40mm option. I then centered the tee, and duplicated it x12 and sliced it. it failed to slice due to an error.

### Actual results

it failed to slice the product

### Expected results

I expected it to slice so I could begin printing

### Add your .zip and screenshots here ⬇️

I don't have a screenshot

GregValiant, 2025-02-05:
Thanks for the report.
I didn't have any trouble slicing it with 4.13.1 so I can't duplicate the problem.
Force the problem again and then use the "Help | Show Configuration Folder" command and find the Cura.log file.  Zip it and post the zip folder here.  Maybe it will have a clue.

Bacteria26, 2025-02-05:
I honestly don’t know wats causing it, so I don’t one how to force the
problem again. It’s been giving me all sorts of issues lately. Mainly just
with printing though. If it shows up again however I’ll follow those steps
and let you know.

GregValiant, 2025-02-06:
4.13.1 might be my favorite version.  I still use it for some things.  Slicing has never been an issue on this very basic Win10Pro laptop.
I'll leave this open.  If it comes up again then post the log file.

Bacteria26, 2025-02-11:
I'm trying to print some golf tees but I can't get them to print. Everytime
I start the print it starts doing its pre-print routine and then after it
touches the top left corner it goes back to the bad, then touches again
then goes to the bottom left doesn't hit, goes back to the pad, then
repeats this twice before disconnecting. Am I doing something wrong? or is
it an issue with the stl file? This only seems to happen when I try to
print golf tees. I've tried 3 different files now all causing the same
issue. It will print 1 out of every 10 tries or so.

GregValiant, 2025-02-11:
Are you printing with the SD card?  It might need to be formatted.

Bacteria26, 2025-02-11:
No I’m printing using the cable

GregValiant, 2025-02-12:
USB Printing has glitches.  The reason it is kept in Cura is for older UM 'Originals' because that's the only way they can print.
I suggest you use an SD card.  If that isn't an option for you I would suggest looking into Octoprint.
If you post one of the problem Gcode files (must be zipped and posted directly to the site and NOT via email) I'll look at it.  It is a very rare thing that the problem is in the Gcode unless it was put there due to problems with the StartUp Gcode which is a definition file or user problem rather than a Cura problem.